respeaker/get_started_with_respeaker #72: Problem when writing data in ReSpeaker Mic Array (Mac)

joansc, 2017-03-02:
Dear ReSpeaker team,

We are having an issue when writing data into the ReSpeaker Mic Array. The Python library cython-hidapi is being used in order to access the registers from the device. We have been able to correctly write and read data following bwhitman's '_listen_and_get_position.py_' script (https://github.com/bwhitman/respeaker-xmos-hid/blob/master/listen_and_get_position.py). However, when writing data and executing the pyhton script few times, the Mic Array hangs/crashes and it doesn't work anymore (doesn't detect any sound, leds not working,...); it has to be disconnected and connected again in order to be able to work well again.

Here I've pasted a simplified script in which an initial value is written to the mic gain register. After few times executing the script the Mic Array doesn't work anymore and has to be reconnected...Any suggestions/ideas/alternatives...?  
Many thanks
Joan

Python script (to execute it just compile in Terminal -> python yourscript.py):

--------

import hid, os

RESPEAKER_VENDOR_ID = 0x2886
RESPEAKER_PRODUCT_ID = 0x07

_dev = hid.device()
_dev.open(RESPEAKER_VENDOR_ID, RESPEAKER_PRODUCT_ID)

def write_register(register, data):
    send_data = [0, register, 0, len(data), 0 ] + data
    return _dev.write(send_data)

def read_register(register, length):
    # To read a register you send reg & 0x80, and then read it back
    # If you have blocking off the read will return none if it's too soon after
    send_data = [0, register, 0x80, length, 0, 0, 0]
    what = _dev.write(send_data)
    ret = _dev.read(len(send_data) + length)
    return ret[4:4+length] # Data comes in at the 4th byte

def main():
    write_register(0x10, [10])
    #mic_gain = read_register(0x10, 1)[0]
    #print "Mic gain is set to %d" % (mic_gain)

if __name__ == '__main__':
    main()

-----

claurier, 2017-03-07:
+1

I am also on MacOS (tried on 10.10 and 10.11).
I tried to update the registry for mic parameters and the audio freezes (no VAD working, no audio stream until I reboot by unplug/plug).
I also tried with node-hid with no luck, it also freezes the audio.
Anyone manages to update the parameters ? Maybe on other platforms ?

thanks !

jerryyip, 2017-03-10:
@claurier @joansc  Will test this.

jerryyip, 2017-03-17:
@claurier @joansc try this: https://github.com/jerryyip/respeaker_micarray_for_mac

joansc, 2017-03-22:
@jerryyip this works when writing on the LEDs, but it freezes when writing values onto sensors (i.e. mic gain)

p-siegel, 2017-06-13:
Same here! I use the c hidapi to interact with the respeaker array.
The first two or three times it works, but then the speaker array crashes and does not even record audio anymore.

I use the read and write functions from [respeaker-xmos-hid](https://github.com/respeaker/respeaker-xmos-hid) to set AGC (0x2A) to off and gain (0x10) to 0.