respeaker/get_started_with_respeaker #73: Is the SD card tutorial from the wiki correct?

burton666, 2017-03-03:
I have tried expanding the filesystem with the included sd-card as instructed in the wiki. I have tried three times without success. First time I lost contact with the respeaker and had to connect putty using COM-port and used some tips in a dd-wrt forum to recover. The other times I also had to connect through serial and run the firstboot -command to recover.

I have used SDFormatter 4.0 to format the card several times and tried again, but same story. I noticed also that last time after formating the card and did the steps in the wiki that when I was supposed to create the second partition using fdisk it complained about thare already was a ext4 partition..

aleifuu, 2017-03-07:
Hi there, i followed the instructions as per README. Seems fine /w me.

Wheels05, 2017-03-08:
I get through most of it fine until Step 4 which shows several things which quide doesnt show but may be correct but cant be sure but then on next step I get

"[ 1301.410000] [EXFAT] trying to mount...
[ 1301.410000] EXT4-fs (mmcblk0p2): couldn't mount as ext3 due to feature incompatibilities
[ 1301.440000] EXT4-fs (mmcblk0p2): couldn't mount as ext2 due to feature incompatibilities
[ 1301.470000] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts: (null)
root@ReSpeaker:/#  couldn't mount as ext3 due to feature incompatibilities"

I copied and pasted pretty much everything to make sure i got it right but still not getting anywhere

burton666, 2017-03-10:
I have copied/pasted every command several times and it have never worked. A couple of times I could fix it by manualy mounting some partitions at the end. But every time I reboot it craches and I am not able to log in using the web.

I always end up having to connect through serial/putty and use mount_root, and firstboot.
I have also never managed to connect to my wifi using the web I always have to connect using serial and use wictl. The respeaker software is far from ready yet. Is there some way to check if the sd-card is genuine if that is the problem?

aleifuu, 2017-03-11:
Maybe it's a long stretch, but from experience brand matters when I was working closely with raspberry pi and other 'firmware' related hardware  ... .
Guys, how 'new' is your microSD Card btw ? and which brand are you using ? I'm using SanDisk and it's been working fine

burton666, 2017-03-11:
I am using the new SanDisk sd-card that was bundle with the respeaker. Btw I tested the sd-card with a program that fills up the entire card then verifies all the data and it was Ok.

aleifuu, 2017-03-12:
well @burton666 , I'm out of ideas at the moment. maybe try firmware update first may resolve it ?

burton666, 2017-03-13:
I guess something is wrong with my unit as the exact copy/paste commands does not produce the result in the guide. And it always loses the wifi info after a reboot.

If I run firstboot this shows up: 
"/dev/mtdblock6 is mounted as /overlay, only erasing files"

How can I fix this? I found some guide for some other hardware where you should enter 
"mount_root"
firstboot
reboot -f

But it still does not fix it.

burton666, 2017-03-13:
Btw. are there any active forum for respeaker that can be used for questions like this or is github the only place to ask questions?

burton666, 2017-03-15:
I am using the latest firmware it says when I try to upgrade.

SpeakUpper123, 2017-04-07:
im having exactly same problems burton666 describes here

solidssss, 2021-03-13:
Same. Broken. Please fix!