respeaker/get_started_with_respeaker #74: Sistem crash

Eternyt, 2017-03-03:
It's happened many times, but I don't know why since I always run the simple playmusic.py programme.
I add the file with the serial output because it's a bit too long to simply copy and paste.
[System crash.txt](https://github.com/respeaker/get_started_with_respeaker/files/817517/System.crash.txt)

jerryyip, 2017-03-06:
Try to disable mopidy&alexa.

Eternyt, 2017-03-06:
I always do that when the system is completely boot up.

jerryyip, 2017-03-06:
not stop. disable them then they will not start again.

Eternyt, 2017-03-06:
Yes but the effects are the same.

aleifuu, 2017-03-07:
Hi @Eternyt, doesnt seem happen to mine. It plays fine. Maybe a lack of power can contribute to that ?

aleifuu, 2017-03-07:
Or maybe you have not execute the right playmusic.py ?

Eternyt, 2017-03-07:
I don't think @aleifuu , because I alway use the same port of my PC and the problem seems random (now it's a while that I don't get this error). The code is the one which I found in the official documentation.

jerryyip, 2017-03-10:
@Eternyt What about updating the firmware?

Eternyt, 2017-03-10:
I update it manually from sd card a few weeks ago. I upload a pictures with the current sistem version.
[image]

jerryyip, 2017-03-10:
And did this happen in other situations?

Eternyt, 2017-03-10:
No, just sometimes when I run the playmusic.py

jerryyip, 2017-03-10:
ok please tell me when it happens again.

Eternyt, 2017-03-10:
Ok. I keep this issue open, if the problem doesn't appear anymore for a while I will close it.

Eternyt, 2017-03-10:
After a while, it happened another time. I was running a slightly modify version of bing_spech.py when the system crash and reboot. Please, notice that I run the same code a moment before without any problem, then I just stopped and re-run it and then the crash occurs. I upload the serial console file, hoping that it can help.
[Sys_crash at 2017-03-10 12.19.19.txt](https://github.com/respeaker/get_started_with_respeaker/files/833768/Sys_crash.at.2017-03-10.12.19.19.txt)

Eternyt, 2017-03-10:
And now it occurred another time!
[Sys_crash at 2017-03-10 17.12.42.txt](https://github.com/respeaker/get_started_with_respeaker/files/834460/Sys_crash.at.2017-03-10.17.12.42.txt)

jerryyip, 2017-03-13:
@Eternyt Have tried your code and it happened on my respeaker too. Need some time to solve it.

KillingJacky, 2017-10-17:
The issue seems be caused by the same root like #120 
@xiongyihui has given a solution there, might not the perfect one, but due to the limitation of the MT7688 chip it's for now the only practical solution.

closed 2017-11-27 by KillingJacky