respeaker/get_started_with_respeaker #76: Default parameters of Mic Array

claurier, 2017-03-09:
I would like to change the default parameters of the Mic Array (like the gain).
What are the options if we want the Mic Array to boot with other default parameters ?
Can we directly change the Mic Array code ?
Or should I do it from the Core (and how) ?

jerryyip, 2017-03-10:
@claurier  You could config it with the HID protocol on the core. Don't need to change the Mic array code.  https://github.com/Fuhua-Chen/ReSpeaker-Microphone-Array-HID-tool

claurier, 2017-03-10:
Thanks @jerryyip !
I had problems doing this from my Mac (Mic array audio stream freezing randomly when changing the values...every 5 attempts or so).
I will try from the Core !

closed 2017-10-17 by KillingJacky