respeaker/get_started_with_respeaker #77: How to control GPIO pins from respeaker

notnatech, 2017-03-10:
Hi,

Any step by step sample on how I can communicate with the GPIO pins of the respeaker?

Thanks

xiongyihui, 2017-03-10:
You can use https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/gpio.py

The example is https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/spi.py, which uses GPIOs to simulate SPI

notnatech, 2017-03-10:
Thanks for the response, how about the pin layout? If I am using mic array, can I just use the pins on the mic array?

Thanks.

elpimous, 2017-03-22:
i d like it too !
could we have expansion header pins legend ?!

elpimous, 2017-03-25:
up please !!

xiongyihui, 2017-03-25:
The pinout is [image]

https://github.com/respeaker/get_started_with_respeaker/blob/master/Introduction.md#hardware

The pinouts of the core and mic array  are the same .

elpimous, 2017-03-25:
thanks a lot, xiongyihui.

elpimous, 2017-03-29:
please, i use vdd5v, ground, lrclk, bclk, and i must use digital out.

what is the dout pin ?? 
i2s_adc ??

thanks
vincent

closed 2017-09-06 by xiongyihui