respeaker/get_started_with_respeaker #78: System freeze on spi.write call

Eternyt, 2017-03-11:
If I call the spi.write in a programme, then the system freeze and I can't do anything else. I must turn off the ReSpeaker and that turn it on again to make it work properly.
This is the code that I use:
[bing_ita.py.zip](https://github.com/respeaker/get_started_with_respeaker/files/835924/bing_ita.py.zip)
And this is a more light version:
[bing_ita2.py.zip](https://github.com/respeaker/get_started_with_respeaker/files/835925/bing_ita2.py.zip)

xiongyihui, 2017-03-12:
Did you stop mopidy and alexa? They can not be used together.

Eternyt, 2017-03-12:
Yes I disabled It a while ago.

joansc, 2017-03-12:
wow, same problem is happening to me when trying to write data using the hiapi library #72 
I also tried the node-hid library (javascript) and got the same error, so I guess there's something wrong on the in-built functions of the device...?
Also, another user mentions a similar error #76

xiongyihui, 2017-03-13:
I wonder if you cloud be able to run `python spi.py` which will try to repeatedly send 'hello' through SPI

Eternyt, 2017-03-13:
I run it, the one in the /usr/lib/python2.7/site-packages/respeaker-0.6.0-py2.7.egg/respeaker folder. And the system doesn't freeze. But I can't verify if the message was properly received by the atmega because your documentation says that spi.write can only send bytearray, and so my arduino code check only the data as a byte.

xiongyihui, 2017-03-14:
How about the following script:

```
from respeaker import spi

while True:
    spi.write(data=bytearray([1, 255, 0, 0]), address=0)
    time.sleep(1)
    spi.write(data=bytearray([1, 0, 255, 0]), address=0)
    time.sleep(1)
```

Eternyt, 2017-03-14:
Yes, it works.

xiongyihui, 2017-03-16:
Then the SPI communication is OK. Do you have any log when the system freezes? Are you able to access it via ssh?

Eternyt, 2017-03-16:
I don't know how to get the system log.
I try with SSH and the system respond, so probably is not a system freeze but a serial port freeze or something similar.
Also, when I run the program and the "system freeze", I connect to it via SSH and I try to run a simple spi writing script: the python works well and I receive all the output ( print('something') ), but I don't get any response from the AtMega. If I reboot and I run the same programme, the Arduino respond.

KillingJacky, 2017-10-17:
Will close this issue, please open a new issue if you have further questions.

closed 2017-10-17 by KillingJacky

Eternyt, 2017-10-18:
Thanks, I will try it. If it will not work i will open a new issue.