respeaker/get_started_with_respeaker #79: Can't setup Wifi

jacksun101, 2017-03-13:
When I first connected into 192.168.100.1, it showed the overview page without asking me wifi network. It seemed logged-in directly, and I can't logout either. So, my wan/internet is always offline. What should I do? I got stuck at beginning.

burton666, 2017-03-13:
If you log in to 192.168.100.1/home you can set up wifi. But if your ssid contains any space it wont work. Then you have to use the terminal.

Eniac1977, 2017-03-13:
Any chance to fix the issue with the space characters in the web page?
Is there any way to automatically configure wifi (e.g. via wictl) when respeaker starts up?

aleifuu, 2017-03-14:
the (temp) solution is to replace space character with "_" ( underscore ) LOL

Mine already saved the information, so when my respeaker starts up, it just automatically connects to the wifi

jerryyip, 2017-03-15:
Hi @Eniac1977  
edit "/etc/config/wireless"

jerryyip, 2017-03-15:
@jacksun101 
Could you connect to the terminal via serial?
If you can, type "firstboot".  https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#factory-reset

SnyperBunny, 2017-03-25:
You could try what I did: I logged into the respeaker network with my cellphone (android) and when it connected it popped up that "login to the network" page that normally pops up for coffee shop wifis. That showed me the wifi setup page and it works well after that. I never could get it to accept wifi information from an actual browser navigating to the /home page. If I try using that, it just "thinks" with a spinning circle until I get sick of waiting and give up.

renval-sanchez, 2017-03-25:
HEY -
got my respeaker yesterday and can't set up wifi either.
my SSID had spaces when I first tried so I changed the wifi name but the new name isn't showing up when I try to connect the respeaker to it.  I've tried on a desktop, iphone, ipad, laptop...  nothing makes any difference.  Any ideas?

renval-sanchez, 2017-03-25:
HEY -
got my respeaker yesterday and can't set up wifi either.
my SSID had spaces when I first tried so I changed the wifi name but the new name isn't showing up when I try to connect the respeaker to it. I've tried on a desktop, iphone, ipad, laptop... nothing makes any difference. Any ideas?

Also, I live in an area with tons of wifi signals...  my other devices show at least 30+ but my resepaker only shows 6-8.  I'm wondering if it's just being overloaded and maxing out.

SnyperBunny, 2017-03-25:
A few suggestions: 
Reboot respeaker (the command is "reboot"). It didn't pick my wifi up last night (I had fiddled around at a friend's house with it and needed to set it back to my wifi when I got home). When it rebooted, mine showed up.

Make sure it has a 2.4Ghz wifi network available to connect to. It does not see my 5Ghz network that most of my devices are connected to.

Setup wifi manually:
https://github.com/respeaker/get_started_with_respeaker/wiki/WiFi
(I don't know 100% how that works but it is worth a try. You will need to be connected via usb and have PUTTY setup to interact with it.)

You could also try going to a place with an open wifi but fewer wifi signals just to test if it will connect there/pickup the signal.

renval-sanchez, 2017-03-25:
Thanks snyperbunny. Noob question, but how do I connect to respeaker to "reboot" it? I've got a Mac. :)

jerryyip, 2017-03-26:
@renval-sanchez 
It is easier to do this on a MAC.
Just connect respeaker to your mac and open terminal.app.
Type "screen /dev/tty.usb" in the terminal and press "tab" key, then you will see something like this:
[image]
Press "enter" key to login respeaker.

jerryyip, 2017-03-26:
Thank you! @SnyperBunny

renval-sanchez, 2017-03-26:
@snyperbunny @jerryyip SUCCESS!!  
Thanks so much!  I'll probably be back soon with more questions but I am indebted to you!! thanks for your help!!

closed 2017-04-16 by jerryyip