respeaker/get_started_with_respeaker #81: System Crash - Red Lights on mic array flash

zdiubaldo, 2017-03-14:
Hi,
I have the latest firmware (Linux ReSpeaker 3.18.23 #83 Sat Nov 19 07:15:53 PST 2016 mips GNU/Linux), connected via serial on OSX, after about 1-2 minutes the mic array leds flash red and the serial terminal is dead and I can't do anything.  I have attached the serial output.  Please let me know what I can do to help debug/fix.  Thanks!

[out.txt](https://github.com/respeaker/get_started_with_respeaker/files/840122/out.txt)

jerryyip, 2017-03-14:
Hi @zdiubaldo ,
Will this happens without Mic array being connected to ReSpeaker?

zdiubaldo, 2017-03-14:
Just tried it without the mic array and got the same result.  Serial console is below.  Is there anything you need to help get this fixed?

[out.txt](https://github.com/respeaker/get_started_with_respeaker/files/841448/out.txt)

burton666, 2017-03-14:
I dont know if I have the same problem but after a while my respeaker crashes or something. Web-ui just throws alot of error message and I have to connect with usb and run firstboot from terminal and set it up again.

zdiubaldo, 2017-03-15:
I am not connected to the web terminal, I am connected the serial terminal the whole time.  That locks up, even if I kill the serial connect and try to connect again, its still locked up.  Only way to recover is to restart the respeaker core.  But then it happens again :)

jerryyip, 2017-03-15:
@burton666 could you post the error message of webui?

zdiubaldo, 2017-03-15:
@jerryyip Yes I am able to ssh into the respeaker once the serial console has locked up.

jerryyip, 2017-03-15:
@zdiubaldo try to update the arduino code https://github.com/respeaker/respeaker_arduino_library/blob/master/examples/pixels_pattern/pixels_pattern.ino

burton666, 2017-03-15:
@jerryyip 
https://s8.postimg.org/4qp5u9rut/Capture.png
https://s8.postimg.org/tl8noccp1/Capture2.png

jerryyip, 2017-03-16:
@burton666 it seems you have closed the http server.
The easiest way to fix this is to flash the firmware again.

zdiubaldo, 2017-03-17:
@jerryyip I was able to keep the respeaker up and not crashing for 1 day.  I updated the pixels_patther.ino like you suggested and updated the ardunio libraries.  That combo seems to fix it.  Feel free to close this issue.

closed 2017-04-10 by jerryyip