respeaker/get_started_with_respeaker #82: Error message after running firstboot

burton666, 2017-03-14:
When I try to reset my respeaker with the command "firstboot" I get this error message : 

"/dev/mtdblock6 is mounted as /overlay, only erasing files"

How do I fix this.

jerryyip, 2017-03-15:
Hi,
You should "umount /dev/mtdblock6" first.

burton666, 2017-03-15:
Ok, I did this and tried again with copy/paste all the commands to add the sd storage. I still don't get the results from df -h as it look in the wiki. 

Here is the log.
[putty_respeaker.txt](https://github.com/respeaker/get_started_with_respeaker/files/843597/putty_respeaker.txt)

burton666, 2017-03-15:
And when I looked again a couple of minutes later I noticed that df -h looks like this:

`Filesystem                Size      Used Available Use% Mounted on
rootfs                    5.2G     11.3M      4.9G   0% /
/dev/root                29.0M     29.0M         0 100% /rom
tmpfs                    61.7M    276.0K     61.5M   0% /tmp
/dev/mmcblk0p2            5.2G     11.3M      4.9G   0% /overlay
overlayfs:/overlay        5.2G     11.3M      4.9G   0% /
tmpfs                   512.0K         0    512.0K   0% /dev
`

jerryyip, 2017-03-15:
@burton666 yes, the storage is extended now.

closed 2017-03-15 by jerryyip

burton666, 2017-03-15:
I disconnected the power and moved the respeaker to another location and after powering up I was not able to connect to it again. Can my hardware be broken?