respeaker/get_started_with_respeaker #83: Issue running the bing.py

jacksun101, 2017-03-15:
I followed the steps on [Programming Guide](https://github.com/respeaker/get_started_with_respeaker/blob/master/ProgrammingGuide.md#how-to-write-a-simple-voice-interaction-program), but it shows error. Anyone can help? Thanks.

- - - - -

INFO:mic:Detected respeaker
Wake up
INFO:mic:Start listening
INFO:requests.packages.urllib3.connectionpool:Resetting dropped connection: speech.platform.bing.com
INFO:mic:Stop listening
1111+100000000-0000000000000000000000000000000000001111+1111111100000000-1111+1111111100000000-000000000000000000000000000000000000000000000
0000000Unexpected response: {u'header': {u'status': u'error', u'properties': {u'NOSPEECH': u'1', u'requestid': u'26801984-1da1-45bf-90f1-749
be62076a4'}}, u'version': u'3.0'}
INFO:mic:Start detecting
INFO:mic:Too many delays, 18 in queue
INFO:mic:Too many delays, 17 in queue
INFO:mic:Too many delays, 16 in queue

xiongyihui, 2017-03-15:
It seems bing speech recognizer didn't recognize any voice. Maybe, the volume is too low. You can use alsamixer to adjust the volume.

jacksun101, 2017-03-15:
Thanks, I just adjusted the alsamixer, and it seems working now. However, the mic is not sensitive, I need to speak out loud to make respeaker hear me. Is it normal?