respeaker/get_started_with_respeaker #84: How do I verify that hardware is Ok?

burton666, 2017-03-15:
I have been having alot of troubles with my respeaker and it basicly stops working a while after adding the sd-card. I have tried multiple times using the command "firstboot", reformating sd-card and copy/paste all the commands from the wiki with the same result. Sometimes it looks like it works but after a while (minutes-hours) it just stops responding and I have to connect it directly to a computer to be able to get access.

How can I verify if this is a software or hardware issue?

jerryyip, 2017-03-16:
Hi @burton666 
Please check your power adapter and see if it provides enough current to respeaker(preferably 1A)

burton666, 2017-03-16:
I have had it powered from a original samsung 2.4A charger and it still locks up.

So what steps do I take to verify this? I have runned firstboot again (for maybe the 20:th time) I will then set up wifi using wictl (As the webui won't ask me for wifi after connecting to 192.168.100.1 and seting up wifi from /home doesn't work either.)

After setting upp wifi I unplugged the respeaker from the computer and moved it to another location and then it dropped out of the wifi again. So as soon as it looses power it resets. Why?

burton666, 2017-03-16:
Ok, I logged in to the respeaker wifi and set it up using wictl using the web-terminal. Should I copy/paste the commands to set up the sd-card again and hope it wont loose settings crash after waiting a while or it reboots? Or should I do something else at first?