respeaker/get_started_with_respeaker #85: Exact steps how to restore respeaker and set up sd-card.

burton666, 2017-03-18:
I have had alot of troubles with my respeaker as it craches and looses the wifi-settings after a short while and the only way to get contact to it again is to connect through COM and set it up again.

I want to restore it again to the exact state it was. Is this correct:
When I connect through com and type "**firstboot**" this comes up: 
"**This will erase all settings and remove any installed packages. Are you sure? [N/y]
y
/dev/mtdblock6 is mounted as /overlay, only erasing files**"
I had to enter this according to respeaker support:
"**umount /dev/mtdblock6**"
When entering "firstboot" again I get this:
"**/dev/mtdblock6 is not mounted
/dev/mtdblock6 will be erased on next mount**"

So after typing "**reboot**" or removing the power everything should be restored after next boot, right?

I will then need to connect through COM again and run "**wictl**" (as my Wifi-SSID will not work using the web-ui)
After wifi is set up I will remove the usb from computer and move respeaker to a wall-charger so that I am 100% sure that power is ok. 
I will then insert the formated bundled sd-card and copy/paste every command from the wiki.

If after all this steps the respeaker stops working, looses wifi setting etc. what can I do?

jerryyip, 2017-03-18:
#84

burton666, 2017-03-18:
I don't know what you mean by that @jerryyip, I have not got any working solutions anywhere earlier.

I just did the first steps again, firstboot.. wicl to set up wifi and then moved it to a wall-charger.
Tried to log in a couple of hours later through the assigned IP and no access again. Wifi-settings have been lost again and I cant access it anyway, and the default Respeaker-wifi is showing again like it has never been configured.

I don't know if the wifi settings where lost directly after I moved it or if it happened later.

So again, How do I fix this??

jerryyip, 2017-03-18:
@burton666 please try to reflash the firmware

burton666, 2017-03-18:
Reflashed firmware and afterreboot I set up wifi, after like a minute of doing nothing this came up in the console:

[  163.090000] capture free_dma_buffer
[  163.100000] ptri2s_config->mmap_index:0
[  163.110000] playback free_dma_buffer
[  163.120000] ptri2s_config->mmap_index:0
[  163.320000] capture free_dma_buffer
[  163.320000] ptri2s_config->mmap_index:0
[  163.340000] playback free_dma_buffer
[  163.350000] ptri2s_config->mmap_index:0
[  163.370000] capture free_dma_buffer
[  163.370000] ptri2s_config->mmap_index:0
[  163.390000] playback free_dma_buffer
[  163.400000] ptri2s_config->mmap_index:0
[  163.680000] ****** wm8960_postinit ******
[  163.700000] playback free_dma_buffer
[  163.710000] ptri2s_config->mmap_index:0
[  163.780000] ummap MMAP[0]=0x820A0000
[  163.790000] playback free_dma_buffer
[  163.800000] ptri2s_config->mmap_index:8

I could still connect putty using both com an ip (respeaker was still connected to computer with usb.)

I moved it to another powerplug, and respeaker lit up three leds green, blue, green and was pulsating those leds like ten times. after a while all leds turned red and then one small green led is turned on and the red led in the middle blinks slowly.
And now I am back to where I started from, nothing works.

jerryyip, 2017-03-26:
Since this question is solved, I will close this ticket.

closed 2017-03-26 by jerryyip

theronic, 2019-02-09:
Is it possible to do a firstboot flash *without* removing the SD card, perhaps by unmounting it first? Every time I take mine out of the case, the button gets damaged.