respeaker/get_started_with_respeaker #86: Install mycroft

burton666, 2017-03-21:
I don't think I ever have managed to solve any task by following this wiki. Last attempt was to install mycroft.

These commands seems to work.
opkg update
opkg install git git-http

next:
git clone https://github.com/respeaker/respeaker_virtualenv.git

First I only got "illegal operation" when I did run the command, then I just rendomly tried to change directory to /home even tho nothing is mentioned about what directory to use. Then I get this:

root@ReSpeaker:/home# git clone https://github.com/respeaker/respeaker_virtualenv.git
Cloning into 'respeaker_virtualenv'...
warning: templates not found /usr/share/git-core/templates
remote: Counting objects: 1438, done.
remote: Compressing objects: 100% (1140/1140), done.
remote: Total 1438 (delta 218), reused 1438 (delta 218), pack-reused 0
Receiving objects: 100% (1438/1438), 4.86 MiB | 337.00 KiB/s, done.
Resolving deltas: 100% (218/218), done.
error: index-pack died of signal 4
fatal: index-pack failed
root@ReSpeaker:/home#

I have no idea if this is ok.

Next:

git clone --depth=2 https://github.com/respeaker/mycroft-core.git

root@ReSpeaker:/home# git clone --depth=2 https://github.com/respeaker/mycroft-core.git
Cloning into 'mycroft-core'...
warning: templates not found /usr/share/git-core/templates
remote: Counting objects: 425, done.
remote: Compressing objects: 100% (284/284), done.
remote: Total 425 (delta 28), reused 349 (delta 13), pack-reused 0
Receiving objects: 100% (425/425), 541.23 KiB | 194.00 KiB/s, done.
Resolving deltas: 100% (28/28), done.
error: unable to write sha1 filename ▒8heR▒@▒:▒▒▒:▒: No such file or directory
fatal: cannot store pack file
fatal: index-pack failed

What is wrong this time?

jerryyip, 2017-03-26:
Hi,
Could you type "df -h" command and post what it returns on here?
It seems respeaker ran out of its flash.

burton666, 2017-03-26:
Here is the output from df -h
[output](http://imgur.com/a/doKos)

burton666, 2017-03-26:
Also the respeaker lights up green every few minutes with green leds lighting up/spinnng and sometimes three leds with blue and green light up.

This comes up in putty if connected through COM

[  464.020000] ****** wm8960_postinit ******
[  464.040000] playback free_dma_buffer
[  464.050000] ptri2s_config->mmap_index:8
[  464.060000] retire_capture_urb: 5 callbacks suppressed
[  473.680000] ummap MMAP[0]=0x81420000
[  473.680000] playback free_dma_buffer
[  473.690000] ptri2s_config->mmap_index:8
[  510.430000] ****** wm8960_postinit ******
[  510.450000] playback free_dma_buffer
[  510.460000] ptri2s_config->mmap_index:8
[  520.080000] ummap MMAP[0]=0x814B0000
[  50.080000] playbak free_dma_buffer
[  520.090000] ptri2s_config->mmap_index:8
[  576.320000] ****** wm8960_postinit ******
[  576.340000] playback free_dma_buffer
[  576.350000] ptri2s_config->mmap_index:8
[  585.970000] ummap MMAP[0]=0x862B0000
[  585.980000] playback free_dma_buffer
[  585.990000] ptri2s_config->mmap_index:8
[  620.400000] ****** wm8960_postinit ******
[  620.420000] playback free_dma_buffer
[  620.430000] ptri2s_config->mmap_index:8
[  630.050000] ummap MMAP[0]=0x81490000
[  630.060000] playback free_dma_buffer
[  630.070000] ptri2s_config->mmap_index:8
[  658.390000] ****** wm8960_postinit ******
[  658.420000] playback free_dma_buffer
[  658.420000] ptri2s_config->mmap_index:8
[  668.040000] umap MMAP[0]=0x81F0000
[  668.050000] plyback free_dma_bffer
[  668.060000] ptri2s_config->mmap_index:8

burton666, 2017-03-26:
And do I need to be in any special directory when issuing the commands?
When I run "git clone https://github.com/respeaker/respeaker_virtualenv.git" from: "root@ReSpeaker:/# " I get " Illegal operation"

I must say that the kickstarter promo made it look like the respeaker was far more userfriendly then this. I think you could atleast update the wiki with exact step by step guides for everything. As there is clearly multiple steps that are missing and will make newbies like me not be able to use the device.

burton666, 2017-03-28:
Any answer to this?

jerryyip, 2017-03-29:
```
[ 464.020000] ****** wm8960_postinit ******
[ 464.040000] playback free_dma_buffer
[ 464.050000] ptri2s_config->mmap_index:8
[ 464.060000] retire_capture_urb: 5 callbacks suppressed
[ 473.680000] ummap MMAP[0]=0x81420000
[ 473.680000] playback free_dma_buffer
[ 473.690000] ptri2s_config->mmap_index:8
```
This comes up because you did not stop alexa and mopidy with following commands:
```
/etc/init.d/mopidy stop
/etc/init.d/alexa stop
```

jerryyip, 2017-03-29:
And it seems your "git" causes the problem above.
So I advise you try this first:
https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#first-impression-with-voice-interaction---respeaker-play-music