respeaker/get_started_with_respeaker #87: MicArray: Are only Left/Right mics working in the new batch?

shinselrobots, 2017-03-21:
Just received a Respeaker Mic Array.  I see 3 LEDs on Left or Right light up as I speak.
When I check it with respeaker-xmos-hid (on Ubuntu), I only see either "90" or "270" degrees reported.

Are only 2 mics enabled?

When I test in a noisy environment, I get poor quality.  Is beam-forming, noise cancellation, etc. enabled? 

Configuration:  Running on Ubuntu 16.04.  Checked on Windows 8.1 as well.

On Windows ReSpeaker Microphone Array Control Panel:
  Revision: v0.31 
  Driver Info: 3.20.0 EVAL  Release Build
  Sample Rate: 16000 Hz
  Output Format: 2 channels, 24 bits
  Buffer Settings:  Safe, Auto

shinselrobots, 2017-03-23:
Hello?  Is anyone from Seeedstudio monitoring these issues?

Fuhua-Chen, 2017-03-24:
Hi，
Can you take a test as below?
1. Download an app that can generate 1kHz audio.
2.Play 1kHz audio and restart the micarray.
3.Do not poweroff the Mic Array, and read the test data in reg address 0x02 via USB HID. 
4.If HID reports  0x7f, it means all mics is well.
5.You can refer to this [HID TOOL](https://github.com/Fuhua-Chen/ReSpeaker-Microphone-Array-HID-tool)

shinselrobots, 2017-03-24:
When I run the test above, I see:

system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ node michid.js set 2 2 0x0000
0 , 0 , 0 , 0 , 0 , 0 , 0 , 0
system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ node michid.js get 2 2
2 , 80 , 2 , 0 , 1 , 8 , 0 , 0
system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ node michid.js get 2 2
2 , 80 , 2 , 0 , 1 , 8 , a5 , 10
system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ node michid.js get 2 2
2 , 80 , 2 , 0 , 1 , 8 , a5 , 10
system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ node michid.js get 2 2
2 , 80 , 2 , 0 , 1 , 8 , a5 , 10
system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ node michid.js get 2 2
/home/system/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool/michid.js:30
        console.log(tmp[0].toString(16),",",tmp[1].toString(16),",",tmp[2].toString(16),",",tmp[3].toString(16),",",tmp[4].toString(16),",",tmp[5].toString(16),",",tmp[6].toString(16),",",tmp[7].toString(16));
                                                                                                                          ^

TypeError: Cannot read property 'toString' of undefined

system@R3-1:~/devel/respeaker/ReSpeaker-Microphone-Array-HID-tool$ 

NOTE:  I see this 'toString'error frequently with this tool.  I don't use node.js, so not sure what is normal.

===========================================
QUESTION:  With the shipped firmware, what is the NORMAL OPERATION?  Should the LED's show the direction of the person's voice, or does the user's software need to read a register to get that data and then set the LED?

What register should I read to get the angle to the user's voice?  Can you send the exact command?

thanks, 
Dave Shinsel
shinselrobots
www.dshinsel.com

shinselrobots, 2017-03-28:
Again,

QUESTION: With the shipped firmware, what is the NORMAL OPERATION? Should the LED's show the direction of the person's voice, or does the user's software need to read a register to get that data and then set the LED?

What register should I read to get the angle to the user's voice? Can you send the exact command?,

jerryyip, 2017-03-29:
Hi,

I think you should try our hid tool on Windows, there are some bugs on Linux.
And the normal operation is the leds are controlled by xmos on mic array and the leds show the direction of the voice.
You are able to send 0x00 0x44 0x80 0x02 0x00 0x00 0x00 0x00 0x00 via usb hid to get the
angle, or mic array will send the angle back automatically per 0.5s.

shinselrobots, 2017-03-29:
Please be specific with your instructions!
what does this mean? "send 0x00 0x44 0x80 0x02 0x00 0x00 0x00 0x00 0x00 via usb hid"

Here is what I tried.  Is this correct?
node michid.js set 0 4 0x4480020000000000
0 , 80 ,  4 , 0 , 0 , 0 , 0 , 0 

node michid.js get 0 4
0 , 80 ,  4 , 0 , 0 , 0 , 0 , 0 

node michid.js get 0 4
0 , 80 ,  4 , 0 , 0 , 0 , 0 , 0 

Always, I get "0 , 80 ,  4 , 0 , 0 , 0 , 0 , 0 "

-------------------------------------------------------------
Also, I tried this other test again, (this time from Windows):
c:\Dev\Respeaker\install\ReSpeaker-Microphone-Array-HID-tool>node michid.js set 2 2 0
2 , 80 , 2 , 0 , 0 , 0 , 0 , 0

c:\Dev\Respeaker\install\ReSpeaker-Microphone-Array-HID-tool>node michid.js get 2 2
2 , 80 , 2 , 0 , 1 , 3f , 0 , 0

if I repeat this test and move the 1kHz sound around, I can get different values, but never higher than 3f
does this mean one microphone is not working?

Also, when I talk, the 3 leds on right or left will light, and *very rarely* the 3 leds in front will light, but the leds in back (near the USB) will never light.

Does this mean I have a bad unit?  Do I need a replacement?  Or is this a known bug?

jerryyip, 2017-03-30:
hi
Sorry for my unclear answer. You could try as the picture shows. 
[image]

|data|degree|
|---|---|
|1e, 0|30|
|e, 1|270|
|d2,0|210|
|96,0|150|

jerryyip, 2017-03-30:
And to test the mic array, 1Khz sound should be loud and close to mic array. Because the test program will run 10 times and there must be more than 8 successes. 
Command to test:
 ```node michid.js get 0x02 2```

shinselrobots, 2017-05-13:
I have continued to try to get this working with no success.  I think I have bad microphones, so I will contact customer support to see if I can get a new board.