respeaker/get_started_with_respeaker #93: no wifi

speeeeeed2006, 2017-03-27:
Hello I have no wifi and no serial using arduino. 
What should I do?

AeroFont, 2017-03-28:
Got the same problem. Turned off all Wifi to test something, wasnt able to turn them on again over console, so I tried firstboot. Did nothing, now the respeaker wont even start, cant even connect to it trough serial anymore. After Months of waiting and finally getting it delivered, this has been a goddamn ride to hell in terms of appreciation of this product.

AeroFont, 2017-03-28:
Ok, after doing the SD card firstboot method, I was able to finally able to connect to it again over WLAN. Now I need to find a way to reset the COM/Serial Port, probably something with the arduino, as the LEDs dont light up anymore

speeeeeed2006, 2017-03-28:
Can you tell me how you made that using the SD card ? 
I used the script in the reset factory section. 
`#!/bin/sh

if [ ! -f /tmp/run/mountd/mmcblk0p1/reset_once ]; then
    firstboot
    touch /tmp/run/mountd/mmcblk0p1/reset_once
    reboot -f
fi`
in a file named `autorun` an put directly in the SD card but it dosen't work.

AeroFont, 2017-03-28:
autorun without file ending did it for me. Just restarted the ReSpeaker again, now its WLAN AP is unavaliable. Quite the bummer, SD Card seems to have no effect now too. Also, dont forget to format the SD before putting the file on it

speeeeeed2006, 2017-03-28:
I formated too used SD Formater and put the file without  file ending.

AeroFont, 2017-03-28:
Its not working for me anymore. Currently preparing for reflashing the atmega, do you got any idea about how to connect an arduino or whatever to do that?

speeeeeed2006, 2017-03-28:
In the wiki they said we can use this https://github.com/respeaker/get_started_with_respeaker/issues/29#issuecomment-277914967
but the problem is that the pin was really small it will be a pain to use them.

AeroFont, 2017-03-28:
I see. Tho I have no idea how to flash this stuff; I am mostly interested in the Python functionality to incorporate in a Project. The small amount I know about the Arduino would be enough to make the lamps light up....

Is there anywhere a Detailed howto abou reseting this whole thing and flashing it? I'd order a new one, but it'd take way too long.

speeeeeed2006, 2017-03-28:
I was also looking for something to flash the respeaker, just like an arduino, but I will wait a little just to have a confirmation.

AeroFont, 2017-03-28:
Ive got my arduino ready with everything connected, Do you know what one would flash on those pins? Some kind of Source or...?

speeeeeed2006, 2017-03-28:
I think you can have some idea using this 
http://hardwarefun.com/tutorials/use-arduino-as-an-isp-programmer-to-program-non-arduino-avr-microcontrollers
or this 
http://www.instructables.com/id/Turn-Your-Arduino-Into-an-ISP/?ALLSTEPS
and use this as a reference 
http://makerdude.com/blog/wp-content/uploads/2012/03/icsp_hookup.png

AeroFont, 2017-03-28:
Done. Ive used this:
https://www.arduino.cc/en/Tutorial/ArduinoISP

To load the LEonardo ETH bootlaoder onto the 32U4. Afterwards the serial port was enables again. Mind yourself, it changed to 9600baud for me. Anyway, I was able to do unmount /dev/XXXXX and then firstboot and reboot

Now its working again.

speeeeeed2006, 2017-03-28:
can you tell me how you done to connect the small pin ?

AeroFont, 2017-03-28:
I soldered wires to it...no other real way to do it I guess.

speeeeeed2006, 2017-03-29:
it haven't worked for me. I think Seeed should made a better wiki for this part because it wasn't very clear.

AeroFont, 2017-03-29:
This is goddamn garbage. The device didnt crash again as it did before, but now the whole Linux/WiFi part doesnt work. I cant acces Juicbox, I cant use wictl to get wifi as blocks the whole system, and using the ip it gets from dhcp doesnt do shit either. 

Quite honestly was any testing done prior to shipping? If by following the guides on here you break half the stuff...

speeeeeed2006, 2017-03-29:
I don't know too I spend a lot of time to fix it but for now it dosn't work.

AeroFont, 2017-03-29:
There seems to be a deeper problem. When doing a complete factory reset and following the steps:
Setup WiFi per 192.168.100.1/home.html
Setup Juicbox to turn off the ReSpeaker AP, setup Time
Insert SD card and format everything
Reboot

Errors:
WLAN AP of the Juicbox turns itself on again
ReSpeaker doesnt reconnect to the previously chosen WLAN. Even after using wictl
Random outages during network communication -> cant access Web anymore, but serial works

speeeeeed2006, 2017-03-29:
I think there's some hardware problems on some respeaker.

AeroFont, 2017-03-31:
Definately. Of probably over 50 Times used, I only got it to work once.
I was able to make it recognize alexa respeaker and play music. After adding some stuff to dictionary and keywords, it broke. 

Also, BING never worked for me. I dont know why but we probably need a FW Update....seeing as 0.99.9 is the newest, there must still be some irks in it.

speeeeeed2006, 2017-03-31:
I still tried last night and I still dosn't work, maybe I do somthing wrong but there's definetly some work to do on software.

speeeeeed2006, 2017-04-03:
no news from seeed for now

jerryyip, 2017-04-07:
@speeeeeed2006 @AeroFont
Was your respeaker insert a sd card? could you pull the sd card out and reboot to see if the wifi works or not?
Thank you!

speeeeeed2006, 2017-04-08:
nothing, no wifi and still no serial.

Ervanu, 2017-05-15:
I have the same problem. Have you got the solution? @speeeeeed2006  @AeroFont

jerryyip, 2017-05-15:
@Ervanu no serial and no wifi? Have you edited something?

Ervanu, 2017-05-15:
No,, I just follow tuttorial on this site bellow:

https://github.com/respeaker/get_started_with_respeaker/wiki/SD-card-autorun-script

is the tutorial in the link above wrong?

Ervanu, 2017-05-15:
Before the problem occurs, the respeaker core can work normally. Alexa and playmusic can running good. I tried to do a program using Arduino IDE. Com Port detected arduino leonardo. But, i can not uploading program to the respeaker core. There was an error with avrdude. Then, I try to follow the steps on the following link:
https://github.com/respeaker/get_started_with_respeaker/wiki/SD-card-autorun-script
but in the middle of procces we suddenly got lost connection with the serial monitor. Then the serial (COM port) and wifi connection wasn't exist anymore

jerryyip, 2017-05-15:
Hi,
what about pulling out the sd card and repower it?

speeeeeed2006, 2017-05-15:
No I still haven't fix it, I was a little busy this days and I haven't touche the Respeaker. But I think i don't have any chance to fix it and using the ICSP pins was the solution but I tried and didn't work ( maybe my solder isn't so good because the pins was small ) so I considering now to buy a new one. 
But why no information about this from Seeed studio, I mean is it a common problem or is it a minor one ?

Ervanu, 2017-05-16:
I tried contacting seeed studio technical support. But, no answer from them. I spent about 1 week to fix this problem. Sometimes wifi is detected but Serial Port never detected again. If wifi can be detected, Respeaker can be acces over ssh console Putty. It can work normally. Web terminal can work as well. I have tried  upgrading Respeaker firmware. Upgrade done. But after reboot, wifi connection broken. Green Led "ON", Blue Led "flash/blink". It take long time about 3-4 hours until wifi can be detected again. Is there are any problem with respeaker hardware?

jerryyip, 2017-05-19:
@Ervanu 
I guess you have messed the original arduino code, which is used for transmitting serial data between mt7688 and your pc. So what you need to do is upload the [example arduino code](https://github.com/respeaker/respeaker_arduino_library/blob/master/examples/pixels_pattern/pixels_pattern.ino) again. Then go to terminal via serial, use "wictl" connect to a Wi-Fi.

Ervanu, 2017-05-22:
Can you tell me about the steps to upload an arduino program (without serial Com Port)? @jerryyip

jerryyip, 2017-05-22:
@Ervanu 
Here should be a serial COM Port of arudino. Are you sure that it can't be found on your computer now?

Ervanu, 2017-05-29:
Yes, Serial COM Port Arduino can't be found on Arduino port / device manager.

KillingJacky, 2017-10-17:
Please refer to #58 `To fix a totally bricked device`

closed 2017-11-27 by KillingJacky