respeaker/get_started_with_respeaker #94: Getting a Strange error BusyBox...

AeroFont, 2017-03-28:
The Error is the following:
ERROR!!! MlmeEnqueueForRecv(): un-recongnized mgmt->subtype=4, STA-1c:b9:c4:07:2a:48

Also, to go ahead and access /root/.pip/pip.log, how do I proceed? As far as I see we dont have FTP on the respeaker...

jerryyip, 2017-03-29:
Hi,
Could you connect via sftp?
And please tell me that when did the error happens?

AeroFont, 2017-03-29:
per sftp I just get an empty folder, ftp doesnt work. The error just appears all the time in the console. Gets posted regularly in it

jerryyip, 2017-03-30:
We don't have ftp but sftp. It works well with my sublime sftp tool and Cyberduck(on mac). Which sftp tool you are using?
And we haven't met the MlmeEnqueueForRecv() error as your post. This may happen when the wifi signal of your router is not good.

AeroFont, 2017-03-31:
Ive found the error. When you power it via Laptop USB (for serial), it doesnt get enough power to maintain a steady connection. Using a normal Power Supply erases the error.

I was able to get onto SFTP afterwards too.

closed 2017-03-31 by AeroFont