respeaker/get_started_with_respeaker #95: IOError: [Errno -9996] Invalid input device (no default output device)

jacksun101, 2017-03-28:
I got error when I ran the command "python bing.py". Please help.

`root@ReSpeaker:~# python bing.py
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "bing.py", line 13, in task
    mic = Microphone(quit_event=quit_event)
  File "/usr/lib/python2.7/site-packages/respeaker-0.6.0-py2.7.egg/respeaker/microphone.py", line 93, in __init__
    input_device_index=self.device_index,
  File "/usr/lib/python2.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno -9996] Invalid input device (no default output device)
`

Eniac1977, 2017-03-28:
you have to kill all processes that use the microphone first (alexa, mopidy)...

jerryyip, 2017-03-29:
Thanks @Eniac1977 , @jacksun101 you could type this first:
```
/etc/init.d/mopidy stop
/etc/init.d/alexa stop
```

I will close this issue as it is solved.

closed 2017-03-29 by jerryyip