respeaker/get_started_with_respeaker #96: audio to text conversion problem

sudhaa1232, 2017-03-28:
Hi team,

I am using following link to start. 
https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md

I executed everything except following
in the python code,
1) I am able to play music but I am not able to read the all the text of what I am speaking. I am trying to do this from linux websocket shell. Is this is correct place? please suggest some way.

2) I also wanted to change wake up command "respeaker" to something else. I replaced word respeaker from             " if mic.wakeup('respeaker'):"            line but it has not worked. 
please suggest solution
  
Thanks
Sudhanshu

Eniac1977, 2017-03-28:
Regarding your second question: You have to define the Keyword in the pocketsphinx configuration. For this, you have to extend the dictionary and the keyword-file. You'll find some HowTos on this website...

jerryyip, 2017-03-29:
1. you could try do connect to respeaker via serial

2. the following link is about how to change a wake up word
https://github.com/respeaker/get_started_with_respeaker/issues/68#issuecomment-285596068

jerryyip, 2017-04-05:
@sudhaa1232  I am going to close this issue as it was solved. Feel free to raise issues.

closed 2017-04-05 by jerryyip