respeaker/get_started_with_respeaker #97: mic array firmware upgrade - os x?

a7ypically, 2017-03-29:
Hi,

I tried using a Windows machine but the dfu uploader fails - it just get stuck after preparing to upload.

Can I use this code for a generic xmos device (after changing the vendor and product id) to upload the firmware from my Mac?
https://github.com/itdaniher/USB-Audio-2.0-Software-v6.1/blob/master/sc_usb_audio/module_dfu/host/xmos_dfu_osx/xmosdfu.cpp

Thanks.

jerryyip, 2017-04-03:
Hi,
We have not test this tool before, but you could try it by yourself.

a7ypically, 2017-04-06:
Is there any risk I will brick the mic if it doesn't work?

jerryyip, 2017-04-07:
No, feel free to upload.
We have tested the tool, it is fine.
[xmosSeeeddfu.cpp.zip](https://github.com/respeaker/get_started_with_respeaker/files/904984/xmosSeeeddfu.cpp.zip)

a7ypically, 2017-04-10:
Works well, thanks.

closed 2017-04-16 by jerryyip