respeaker/get_started_with_respeaker #102: wifi config, switch from 'ap' to 'sta' mode

smkaeser, 2017-04-04:
when the 'ap' interface is enabled and the 'sta' interface is disabled in the /etc/config/wireless the respeaker connects to my home wifi and I can access it over ssh (also when I am not connected directly to the wifi of the respeaker). However when I disable 'ap' and enable 'sta' the respeaker does not connect to the home wifi anymore after a reboot and I have to switch the configuration back over usb. Thanks for any hint on this issue!

jerryyip, 2017-04-06:
hi,
please try `wictl` command to connect to the home wifi.
[image]

smkaeser, 2017-04-09:
Thanks for you answer. I already tried this. Even when connecting to the wifi with wictl, the access point "ReSpeakerXXXXX" stays activated. I would like to turn off the access point and operate the ReSpeaker in station mode. How do I do that once I am connected to the home wifi.

the config/wireless file looks like this after wictl. what do I have to change to turn the access point of and station mode on?
config wifi-iface 'ap'
        option device 'radio0'
        option mode 'ap'
        option network 'lan'
        option ifname 'ra0'
        option ssid 'ReSpeaker1DD5DE'
        option seq '1'
        option key '*******'
        option encryption 'psk2'

config wifi-iface 'sta'
        option device 'radio0'
        option mode 'sta'
        option network 'wan'
        option ifname 'apcli0'
        option led 'mediatek:orange:wifi'
        option ssid 'UplinkAp'
        option key 'SecretKey'
        option encryption 'psk'
        option disabled '1'
        option ApCliEnable '1'
        option ApCliAuthMode 'WPA2PSK'
        option ApCliSsid '****'
        option ApCliWPAPSK '******'
I tried to set the "disable" flag of the 'ap' interface to '0' and the one of the 'sta' interface to '1'. But then the ReSpeaker did not connect to the wifi anymore.

jerryyip, 2017-04-11:
Hi
You could set the 'ssid' flag of the 'ap' interface to ''
Then the ap will be hidden.

jerryyip, 2017-04-16:
I will close this issue, thanks.

closed 2017-04-16 by jerryyip

theronic, 2019-02-12:
Does the /etc/config/wireless config file pertain to wictl, or to some other service like orange?