respeaker/get_started_with_respeaker #104: Mic Array & SD Card issue

jacksun101, 2017-04-06:
I've got a new Mic Array recently, and it works without sd card inserted into the Core. However, if I inserted the sd card and reboot the respeaker. It just doesn't seem to work at all. I can not see the wifi ssid, nor ssh login. Even the web ui is not accessible.

jerryyip, 2017-04-07:
Hi,
Does this problem happen every time? It never happens on my respeaker and mic array. 
And could you post the boot log?

Thank you!

jacksun101, 2017-04-07:
@jerryyip This issue happens all the time when I use Core + MicArray + SD card together. If I remove the SD Card it works as normal. The log seems only have the current session (with the sd card removed), and it looks nothing special too me.

> Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: Got a autofs packet
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: kernel is requesting a mount -> mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: request for invalid path /tmp/run/mountd/mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: failed to mount mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: Got a autofs packet
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: kernel is requesting a mount -> mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: request for invalid path /tmp/run/mountd/mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: failed to mount mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: Got a autofs packet
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: kernel is requesting a mount -> mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: request for invalid path /tmp/run/mountd/mmcblk0p1
Fri Apr  7 22:51:27 2017 user.crit mountd[1674]: failed to mount mmcblk0p1
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.220000] capture free_dma_buffer
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.230000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.240000] playback free_dma_buffer
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.250000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.440000] capture free_dma_buffer
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.450000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.470000] playback free_dma_buffer
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.480000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.490000] capture free_dma_buffer
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.500000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.520000] playback free_dma_buffer
Fri Apr  7 22:51:31 2017 kern.warn kernel: [  112.520000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:32 2017 kern.warn kernel: [  112.800000] ****** wm8960_postinit ******
Fri Apr  7 22:51:32 2017 kern.warn kernel: [  112.820000] playback free_dma_buffer
Fri Apr  7 22:51:32 2017 kern.warn kernel: [  112.830000] ptri2s_config->mmap_index:0
Fri Apr  7 22:51:32 2017 kern.warn kernel: [  112.900000] ummap MMAP[0]=0x82000000
Fri Apr  7 22:51:32 2017 kern.warn kernel: [  112.910000] playback free_dma_buffer
Fri Apr  7 22:51:32 2017 kern.warn kernel: [  112.920000] ptri2s_config->mmap_index:8
Fri Apr  7 22:54:40 2017 auth.err sshd[2081]: error: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Fri Apr  7 22:54:40 2017 auth.err sshd[2081]: error: Could not load host key: /etc/ssh/ssh_host_ed25519_key
Fri Apr  7 22:54:40 2017 auth.info sshd[2081]: WARNING: /etc/ssh/moduli does not exist, using fixed modulus
Fri Apr  7 22:54:45 2017 auth.info sshd[2081]: Accepted password for root from 192.168.31.199 port 1834 ssh2
Fri Apr  7 22:54:45 2017 authpriv.info syslog: pam_unix(sshd:session): session opened for user root by (uid=0)
Fri Apr  7 22:54:45 2017 auth.info sshd[2081]: WARNING: /etc/ssh/moduli does not exist, using fixed modulus
Fri Apr  7 22:54:45 2017 auth.err sshd[2085]: error: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Fri Apr  7 22:54:45 2017 auth.err sshd[2085]: error: Could not load host key: /etc/ssh/ssh_host_ed25519_key
Fri Apr  7 22:54:45 2017 auth.info sshd[2085]: WARNING: /etc/ssh/moduli does not exist, using fixed modulus
Fri Apr  7 22:54:47 2017 auth.info sshd[2085]: Accepted password for root from 192.168.31.199 port 1842 ssh2
Fri Apr  7 22:54:47 2017 authpriv.info syslog: pam_unix(sshd:session): session opened for user root by (uid=0)

jacksun101, 2017-04-07:
OK, I've found the reason. It's because the USB connector. I was using the USB connector which is not thick enough. Now I am using the USB connector that Respeaker provides, and it works. God, I hate myself. Let's close this issue now. Thanks @jerryyip

jacksun101, 2017-04-07:
Well, the issue happens again and again. I removed the cover of Pro Case, sometimes it works and sometimes doesn't.

jerryyip, 2017-04-10:
How about removing the whole Pro Case? Was it too hot?

jacksun101, 2017-04-10:
@jerryyip Removing the case doesn't help. It's nothing related to the MicArray now. Even the Core can't reboot successfully with inserted sd card. At the same time, I am working very hard on the new build for myself. For the past few weeks, I realized there are so many issues need to tackle to develop my project using Respeaker.

jacksun101, 2017-04-10:
The following logs are captures from serial console. The boot process hangs there with sd card inserted into the Respeaker Core.

> [    6.960000] random: procd urandom read with 11 bits of entropy available
[    6.980000] platform gpio-leds: Driver leds-gpio requests probe deferral
[    7.370000] rt305x-esw 10110000.esw: link changed 0x00
Press the [f] key and hit [enter] to enter failsafe mode
Press the [1], [2], [3] or [4] key and hit [enter] to select the debug level
[   10.760000] mount_root: loading kmods from internal overlay
[   11.210000] jffs2: notice: (377) jffs2_build_xattr_subsystem: complete building xattr subsystem, 0 of xdatum (0 unchecked, 0 orphan) and 0 of xref (0 dead, 0 orphan) found.
[   11.240000] block: attempting to load /tmp/jffs_cfg/upper/etc/config/fstab
[   11.350000] EXT4-fs (mmcblk0p2): recovery complete
[   11.360000] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Opts:
[   11.390000] mount_root: switched to extroot
Illegal instruction
[   11.440000] watchdog watchdog0: watchdog did not stop!
[   11.450000] Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   11.450000]
[   11.450000] ---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
[   11.450000]
[   69.720000] random: nonblocking pool is initialized

jacksun101, 2017-04-10:
Ok, it turned out that sd card was broken. After a low-level format of the sd card, and issue "firstboot" command, then restart all the sd card process, it works now.

closed 2017-04-10 by jacksun101