respeaker/get_started_with_respeaker #106: ssh access denied

jacksun101, 2017-04-09:
I've spent quite amount of time to rebuild the openwrt bin file for respeaker and it's success finally. However, after upgraded to the new build, there is no dashboard on the WebUI and I can't ssh login in either using root id/password. Any suggestion?

> login as: root
Access denied
Using keyboard-interactive authentication.
Password:
Access denied
Using keyboard-interactive authentication.
Password:

xiongyihui, 2017-04-10:
Did you have your sd card inserted? Upgrade will not reset the contents of the sd card.
Maybe remove `/root/.ssh`

jacksun101, 2017-04-10:
Thanks, @xiongyihui I use serial console, now I can login as root. For others meet the situation like me, please connect your respeaker to a Mac and find your usb identity string, then issue the command as follow.

screen /dev/xxx 57600

closed 2017-04-10 by jacksun101

jerryyip, 2017-04-10:
1. You could login via serial and reflash the firmware. [Guide](https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#update-for-old-version)
2. have you chose the package of webui when you rebuilt?

jacksun101, 2017-04-10:
@jerryyip Yes, I've reflash the official firmware. However, for the new firmware I've built, I did choose the webui package, but it doesn't have dashboard, and some other pages.