respeaker/get_started_with_respeaker #107: stuck with recover

pastukhov, 2017-04-09:
Hello!
My respeaker is not connecting anymore to wifi and when i'm tying to get serial console with putty or screen, i'm getting ???? or garbage on any key press.

Can anyone help me?

jacksun101, 2017-04-10:
I use respeaker serial console with my mac, and it works. I had bad experience with those low quality USB connector, so please use the connector that respeaker provide. And try the following command, and replace the usb identity string to yours.

screen /dev/tty.usbmodem1411 57600

jerryyip, 2017-04-10:
@pastukhov 
Try to reflash the arduino code [here](https://github.com/respeaker/respeaker_arduino_library/blob/master/examples/pixels_pattern/pixels_pattern.ino). Because the uart of mt7688 is connected to atmega 32u4 and 32u4 is a intermediator between mt7688 and your pc.

pastukhov, 2017-04-12:
Thanks @jerryyip!
It's working after pixels_pattern flashing.

closed 2017-04-12 by pastukhov