respeaker/get_started_with_respeaker #109: Invalid number of channels when using Alexa

brucechu777, 2017-04-14:
when I run alexa, I got the error below
get error IOError: [Errno -9998] Invalid number of channels

so I stop mopidy , but when I run etc/init.d/mopidy stop.  I also got the same error

jerryyip, 2017-04-16:
Hi,
could you please reboot your respeaker and try this again?

brucechu777, 2017-04-17:
I have tried several times ,but it did not work.

root@ReSpeaker:~# mopidy -o section/key=stop
INFO     Starting Mopidy 2.0.1
INFO     Loading config from builtin defaults
INFO     Loading config from /root/.config/mopidy/mopidy.conf
INFO     Loading config from command line options
INFO     Enabled extensions: hallo, mpd, alsamixer, http, stream, m3u, filemanager, softwaremixer, file, musicbox_webclient, local
INFO     Disabled extensions: none
INFO     Starting Mopidy mixer: AlsaMixer
INFO     Mixing using ALSA, card 0, mixer control "Headphone".
INFO     Starting Mopidy audio
INFO     Starting Mopidy backends: StreamBackend, M3UBackend, FileBackend, LocalBackend
INFO     Audio output set to "autoaudiosink"
INFO     No local library metadata cache found at /root/.local/share/mopidy/local/library.json.gz. Please run `mopidy local scan` t
o index your local music library. If you do not have a local music collection, you can disable the local backend to hide this messa
ge.
INFO     Loaded 0 local tracks using json
INFO     Starting Mopidy core
INFO     Starting Mopidy frontends: MpdFrontend, HttpFrontend, HalloFrontend
INFO     MPD server running at [::ffff:127.0.0.1]:6600
INFO     HTTP server running at [::ffff:0.0.0.0]:6680
INFO     Starting GLib mainloop
INFO     New MPD connection from [::ffff:127.0.0.1]:42908
Expression 'parameters->channelCount <= maxChans' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 1513
Expression 'ValidateParameters( inputParameters, hostApi, StreamDirection_In )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line:
 2813
Exception in thread Thread-13:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/site-packages/mopidy_hallo/__init__.py", line 29, in _run
    mic = Microphone(quit_event=self.quit_event)
  File "/usr/lib/python2.7/site-packages/respeaker/microphone.py", line 89, in __init__
    input_device_index=self.device_index,
  File "/usr/lib/python2.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/lib/python2.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
IOError: [Errno -9998] Invalid number of channels

xyu6, 2017-11-24:
There is another application or alexa instance using the audio input device. Run /etc/init.d/alexa stop and /etc/init.d/mopidy stop to stop them. To disable mopidy to startup, run /etc/init.d/mopidy disable.

closed 2017-12-15 by KillingJacky