respeaker/get_started_with_respeaker #110: Mic array - noisy input

muddygs, 2017-04-14:
Setup: Win10, ReSpeaker v0.32, driver 3.20.0 EVAL
[respeaker-mic-noise-sample.zip](https://github.com/respeaker/get_started_with_respeaker/files/922528/respeaker-mic-noise-sample.zip)

I've had this mic working once, but usually when I plug it in, I get what essentially sounds like static. Three of the LEDs keep flashing. Angles are almost always reports as:
45 , 80 , 2 , 0 , d2 , 0 , 0 , 0 or 45 , 80 , 2 , 0 , e , 1 , 0 , 0.

When I say noisy, when I start up in a silent room, the output from the mic is similar to static. Seem to get a similar result in a noisy environment.

Please let me know if you need further info or what I should try next. Thank you.

ms4246, 2017-04-15:
I'm also just getting static without any voice. file: https://ufile.io/57aec

shyhshian, 2017-05-18:
I found mic noisy issue was due to some mics (1 or more of the 7-mics) have mic_data Jitter issue (may be caused by waveform ripple). the mic_data may not be aligned to mic_clock, so XMOS got wrong audio data. I add 58pF capacitor connect from mic_data to GND (you can see, there are spared pads near microphone). the issue solved. For your reference!

takaomoriyama, 2017-10-19:
@shyhshian According to your suggestion, I did soldering of 56pF capacitors to my array microphone board, and now It works pretty fine.

Fuhua-Chen, 2017-10-19:
Please try to add cap as above users did, if still not work well. try to remove cap C87,C94,C101,C103,C104,C105,C106

closed 2017-11-27 by KillingJacky