respeaker/get_started_with_respeaker #111: crash at network start

Ronthecat, 2017-04-17:
See attached file. Hard crash running /etc/init.d/network, I renamed /etc/init.d/network to /etc/init.d-network to prevent it from running and the system boots up and runs, eth0 is visible but no wifi. If I try to start the network service it crashes with the same final messages. I tried factory reset, I flashed the latest .bin from the command line and sdcard. 

> [   28.630000] br-lan: port 1(eth0.1) entered forwarding state
> [   30.250000] efuse_probe: efuse = 10000012
> [   30.890000] net ra0: Direct firmware load for mt7628.eeprom failed with error -2
> [   30.910000] net ra0: Falling back to user helper

[Respeaker-output-April17th2017-1.txt](https://github.com/respeaker/get_started_with_respeaker/files/924973/Respeaker-output-April17th2017-1.txt)

When I tried to boot with option 5 to update system from sdcard, it gave me "Bad Magic Number,9C65F91D", that's all it says when selecting any option 1-7.

[Respeaker-Boot-Options-April17th2017-1.txt](https://github.com/respeaker/get_started_with_respeaker/files/924980/Respeaker-Boot-Options-April17th2017-1.txt)

jerryyip, 2017-04-25:
Hi,
Please reflash the firmware and see if it happens again.
[Reflash firmware toturial here.](https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#update-for-old-version)

Ronthecat, 2017-04-25:
I did, twice... Same thing, start the wire less and it locks up hard, no
serial console, no nothing. And the not menu options do not let me boot to
load over the network or SD card. I had to disable the networking as soon
as I could (serial terminal) before it turned the wifi on and locked up.
Then I was able to do the firmware upgrade from the SD card. But it did not
change anything. Hardware issue? It's been a long time I've been waiting
for it to show up...

Ronthecat, 2017-04-27:
Is this a hardware problem? Waiting for a long time...

Flashed
firstboot
Flashed

Every time it starts the wifi, it locks up hard.

You did not even read my comment...  

>  I tried factory reset, I flashed the latest .bin from the command line and sdcard.

None of the boot options work, I can not update over ethernet, none of the options work...

jerryyip, 2017-04-28:
@Ronthecat I am sorry to hear that. If you bought this board from KS, email to lanqing.mei@seeed.cc   or email to order@seeed.cc if you bought from seeed barzaar.
You could apply for replacement.

closed 2017-10-17 by KillingJacky