respeaker/get_started_with_respeaker #113: Weird Start with Putty (Windows)

wuchouk, 2017-04-20:
There are two situations when I connected ReSpeaker with Putty. 
What should I do so that I can access properly?

Thanks!

1. It won't show anything, it just wait there. (it might stuck at the end of the second situation)

2. Pop out these information and stop at the autorun

OK
No initrd
## Transferring control to Linux (at address 80000000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 3.18.23 (yihui@seserver) (gcc version 4.8.3 (OpenWr                                                                                        t/Linaro GCC 4.8-2014.04 r48532) ) #8 Tue Nov 22 16:39:11 CST 2016
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
[    0.000000] MIPS: machine is Seeed ReSpeaker v1 7688
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x07ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 byt                                                                                        es
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pag                                                                                        es: 32512
[    0.000000] Kernel command line: console=ttyS2,57600 rootfstype=squashfs,jffs                                                                                        2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=000525a4
[    0.000000] Readback ErrCtl register=000525a4
[    0.000000] Memory: 126272K/131072K available (2551K kernel code, 128K rwdata                                                                                        , 524K rodata, 180K init, 186K bss, 4800K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:256
[    0.000000] intc: using register map from devicetree
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource_of_init: no matching clocksources found
[    0.000000] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.080000] pinctrl core: initialized pinctrl subsystem
[    0.090000] NET: Registered protocol family 16
[    0.110000] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.120000] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.130000] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.140000] Switched to clocksource MIPS
[    0.150000] NET: Registered protocol family 2
[    0.150000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.170000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.180000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.190000] TCP: reno registered
[    0.200000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.210000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.220000] NET: Registered protocol family 1
[    0.230000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.250000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.260000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORIT                                                                                        Y) (c) 2001-2006 Red Hat, Inc.
[    0.290000] msgmni has been set to 246
[... 127 lines cut ...]
[   28.460000] br-lan: port 1(eth0.1) entered forwarding state
[   28.470000] br-lan: port 1(eth0.1) entered forwarding state
[   29.740000] efuse_probe: efuse = 10000012
[   30.380000] tssi_0_target_pwr_g_band = 34
[   30.390000] tssi_1_target_pwr_g_band = 35
[   30.470000] br-lan: port 1(eth0.1) entered forwarding state
[   37.190000] random: nonblocking pool is initialized
[   37.710000] <==== rt28xx_init, Status=0
[   39.110000] device ra0 entered promiscuous mode
[   39.120000] br-lan: port 2(ra0) entered forwarding state
[   39.130000] br-lan: port 2(ra0) entered forwarding state
[   41.130000] br-lan: port 2(ra0) entered forwarding state
[   42.140000] FAT-fs (mmcblk0p1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
autorun

---------

I've tried firstboot, cuz I thought it might be something wrong, but it only show this

root@ReSpeaker:~# firstboot
This will erase all settings and remove any installed packages. Are you sure? [N/y]
y
/dev/mtdblock6 is mounted as /overlay, only erasing files

thanks

jerryyip, 2017-04-25:
1. What did you mean about "It won't show anything" ? You could type firstboot at command line. That was what the respeaker shows. Note that respeaker doesn't have a desktop or UI, just command line.
2. After "firstboot" and typing "y", respeaker will be back to factory when you reboot it.

wuchouk, 2017-04-27:
hi @jerryyip 

thanks so much for your reply! 

I somehow fixed the first issue, so don't worry on that one anymore.

As for the reboot, when I see this message "/dev/mtdblock6 is mounted as /overlay, only erasing files"
I thought it only erases the new file I uploaded but not resetting the device back to factory setting. 
I have not idea if my device is setting back to factory or not.

But anyway, I have difficulty using this device and decided to leave it first.
Thanks for the help!

closed 2017-10-17 by KillingJacky