respeaker/get_started_with_respeaker #117: DEVELOPERS WHERE ARE YOU?

Eternyt, 2017-04-29:
I'm not happy to spam the issues page with this but is the only place where (hopefully) the devs will read this. I have four open issue opened months ago and nobody answered me!!! And other like me!

jerryyip, 2017-05-03:
Hi friend,
We are sorry for our late reply, but we have answered your question already. If you are not satisfied with our reply, hope you tell us what you need or consult to the other guys here. 
Thanks!

Eternyt, 2017-05-04:
https://github.com/respeaker/get_started_with_respeaker/issues/74 I don't think that is answered...

KillingJacky, 2017-10-17:
You need to post the proper issue thread so we can assist you better. Creating an issue such as this is not helpful to solving issue #74 .

closed 2017-10-17 by KillingJacky