respeaker/get_started_with_respeaker #118: Could I flash image through another way ( such as SD card) ?

iamblue, 2017-05-03:
HI, I flash your new image (https://s3-us-west-2.amazonaws.com/respeaker.io/firmware/ramips-openwrt-latest-LinkIt7688-squashfs-sysupgrade.bin)  to respeaker, but respeaker cannot boot normally, I can't use terminal for to flash new image again... 

So, could you have another way for flashing image ? 

Following is my booting log:

```
flash manufacture id: ef, device id 40 19
find flash: W25Q256FOK
No initrd
## Transferring control to Linux (at address 80000000) ...
## Giving linux memsize in MB, 128

Starting kernel ...

[    0.000000] Linux version 3.18.23 (pillar@server) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.04 r49389) ) #83 Sat Nov 19 07:15:53 PST 2016
[    0.000000] Board has DDR2
[    0.000000] Analog PMU set to hw control
[    0.000000] Digital PMU set to hw control
[    0.000000] SoC Type: MediaTek MT7688 ver:1 eco:2
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019655 (MIPS 24KEc)
[    0.000000] MIPS: machine is Seeed ReSpeaker v1 7688
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 08000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x00000000-0x07ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x00000000-0x07ffffff]
[    0.000000] Initmem setup node 0 [mem 0x00000000-0x07ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 32512
[    0.000000] Kernel command line: console=ttyS2,57600 rootfstype=squashfs,jffs2
[    0.000000] PID hash table entries: 512 (order: -1, 2048 bytes)
[    0.000000] Dentry cache hash table entries: 16384 (order: 4, 65536 bytes)
[    0.000000] Inode-cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Writing ErrCtl register=00012920
[    0.000000] Readback ErrCtl register=00012920
[    0.000000] Memory: 126272K/131072K available (2551K kernel code, 128K rwdata, 524K rodata, 180K init, 186K bss, 4800K reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] NR_IRQS:256
[    0.000000] intc: using register map from devicetree
[    0.000000] CPU Clock: 580MHz
[    0.000000] clocksource_of_init: no matching clocksources found
[    0.000000] Calibrating delay loop... 385.84 BogoMIPS (lpj=1929216)
[    0.060000] pid_max: default: 32768 minimum: 301
[    0.060000] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.070000] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.080000] pinctrl core: initialized pinctrl subsystem
[    0.090000] NET: Registered protocol family 16
[    0.110000] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.120000] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.130000] mt7621_gpio 10000600.gpio: registering 32 gpios
[    0.140000] Switched to clocksource MIPS
[    0.150000] NET: Registered protocol family 2
[    0.150000] TCP established hash table entries: 1024 (order: 0, 4096 bytes)
[    0.170000] TCP bind hash table entries: 1024 (order: 0, 4096 bytes)
[    0.180000] TCP: Hash tables configured (established 1024 bind 1024)
[    0.190000] TCP: reno registered
[    0.200000] UDP hash table entries: 256 (order: 0, 4096 bytes)
[    0.210000] UDP-Lite hash table entries: 256 (order: 0, 4096 bytes)
[    0.220000] NET: Registered protocol family 1
[    0.230000] futex hash table entries: 256 (order: -1, 3072 bytes)
[    0.250000] squashfs: version 4.0 (2009/01/31) Phillip Lougher
[    0.260000] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
[    0.290000] msgmni has been set to 246
[    0.290000] io scheduler noop registered
[    0.300000] io scheduler deadline registered (default)
[    0.310000] Serial: 8250/16550 driver, 4 ports, IRQ sharing disabled
[    0.330000] 10000c00.uartlite: ttyS0 at MMIO 0x10000c00 (irq = 28, base_baud = 2500000) is a 16550A
[    0.340000] 10000d00.uart1: ttyS1 at MMIO 0x10000d00 (irq = 29, base_baud = 2500000) is a 16550A
[    0.360000] console [ttyS2] disabled
[    0.370000] 10000e00.uart2: ttyS2 at MMIO 0x10000e00 (irq = 30, base_baud = 2500000) is a 16550A
[    0.390000] console [ttyS2] enabled
[    0.390000] console [ttyS2] enabled
[    0.400000] bootconsole [early0] disabled
[    0.400000] bootconsole [early0] disabled
[    0.420000] spi-mt7621 10000b00.spi: sys_freq: 193333333
[    0.440000] m25p80 spi32766.0: found w25q256, expected mx25l25635e
[    0.450000] m25p80 spi32766.0: w25q256 (32768 Kbytes)
[    0.460000] m25p80 spi32766.0: using chunked io
[    0.470000] 4 ofpart partitions found on MTD device spi32766.0
[    0.480000] Creating 4 MTD partitions on "spi32766.0":
[    0.490000] 0x000000000000-0x000000030000 : "u-boot"
[    0.510000] 0x000000030000-0x000000040000 : "u-boot-env"
[    0.520000] 0x000000040000-0x000000050000 : "factory"
[    0.530000] 0x000000050000-0x000002000000 : "firmware"
[    0.610000] 2 uimage-fw partitions found on MTD device firmware
[    0.620000] 0x000000050000-0x000000169abb : "kernel"
[    0.630000] 0x000000169abb-0x000002000000 : "rootfs"
[    0.640000] mtd: device 5 (rootfs) set to be root filesystem
[    0.650000] 1 squashfs-split partitions found on MTD device rootfs
[    0.660000] 0x000001e40000-0x000002000000 : "rootfs_data"
[    0.690000] ralink_soc_eth 10100000.ethernet eth0: ralink at 0xb0100000, irq 5
[    0.700000] mt7621_wdt 10000120.watchdog: Initialized
[    0.710000] TCP: cubic registered
[    0.720000] NET: Registered protocol family 17
[    0.730000] bridge: automatic filtering via arp/ip/ip6tables has been deprecated. Update your scripts to load br_netfilter if you need this.
[    0.750000] 8021q: 802.1Q VLAN Support v1.8
[    0.770000] squashfs: SQUASHFS error: unable to read id index table
[    0.780000] jffs2: Flash size not aligned to erasesize, reducing to 31296KiB
[    0.790000] jffs2: jffs2_scan_eraseblock(): Magic bitmask 0x1985 not found at 0x00000000: 0x7368 instead
```

iamblue, 2017-05-03:
I fixed this issue , thx!

petermortensen007, 2017-05-07:
I got the same issue after updating software. I get a kernel panic after all these JFFS2 errors. How did you recover it?

iamblue, 2017-05-27:
@petermortensen007  plz see here: 
https://github.com/respeaker/get_started_with_respeaker/wiki/Rescue-from-a-failed-upgrade

closed 2017-10-17 by KillingJacky