respeaker/get_started_with_respeaker #120: Getting "Illegal instruction" & "Segmentation fault" errors

Oliv4945, 2017-05-08:
Hi,
In lot of softwares (Python, Git, vi...) I am getting those errors, usually `Illegal instruction`, start the software again then `Segmentation fault`, then it works. But sometime only one...
Could you please help ? It is really annoying. 
I tried to erase the SD card and reinstall the Respeaker thanks to `firstboot` without success.

```
root@ReSpeaker:~/jarvis# git pull
Illegal instruction
root@ReSpeaker:~/jarvis# git pull
Segmentation fault
root@ReSpeaker:~/jarvis# git pull
remote: Counting objects: 90, done.
remote: Compressing objects: 100% (43/43), done.
remote: Total 90 (delta 59), reused 73 (delta 42), pack-reused 0
Unpacking objects: 100% (90/90), done.
From https://github.com/oliv4945/jarvis
   36f9f2f..cef1d65  respeaker  -> origin/respeaker
root@ReSpeaker:~/jarvis# 
```

jerryyip, 2017-05-15:
Hi, 
Does it still happen?

Oliv4945, 2017-05-15:
Hi, it still happens and make the Repeaker almost useless in lot of cases :/
A friend of mine have those errors too, but fewer occurrences.

Nothing appears in the log, so if you have any idea ?

Also, it seems that [this issue](https://github.com/respeaker/get_started_with_respeaker/issues/39#issuecomment-280805570) has the same problem, it says `illegal instruction` the first time, then works on second try.

Oliv4945, 2017-05-15:
Other issues related to occasional segmentation fault. [#16](https://github.com/respeaker/get_started_with_respeaker/issues/16), [#13](https://github.com/respeaker/get_started_with_respeaker/issues/13#issuecomment-296881729) and [#61](https://github.com/respeaker/get_started_with_respeaker/issues/61#issuecomment-282465909)

Noxeus, 2017-05-15:
I have this too, a lot.

Oliv4945, 2017-06-07:
Hi @jerryip, any idea ?
Both of my Respeaker are not reliable because of this issue, so they are useless

Oliv4945, 2017-06-30:
@jerryyip : Hi, two month without any update... I can not use my Respeakers please help

@Noxeus : Do you still have the issue ?

jerryyip, 2017-07-02:
@Oliv4945 have you extend the storage with the sd card? https://github.com/respeaker/get_started_with_respeaker/blob/master/QuickStart.md#use-sd-card-to-extend-storage

Oliv4945, 2017-07-02:
@jerryyip Yes, I did. A friend of mine who has the same issue did it too.

Noxeus, 2017-07-03:
@Oliv4945 Yes I still have the issue. Respeaker is unusable for any tasks.

kelnos, 2017-09-10:
Any updates here?  I have this issue as well, running gcc, gdb, python apps (including mopidy), even a simple "hello world" C app sometimes.  It's very unpredictable, sometimes happens nearly immediately after starting a program, sometimes the program works for a little while before it crashes.

This is very concerning; either there's something wrong with how the entire system is built, or the hardware itself is incredibly unstable.  Either way, this is making my ReSpeaker basically useless and I'm strongly considering trying to see if returning it is possible.

Debugging seems to be not all that useful either (though perhaps I need to do some more work to set up gdb?):

```
$ ./build_dir/toolchain-mipsel_24kec+dsp_gcc-4.8-linaro_uClibc-0.9.33.2/gdb-linaro-7.6-2013.05/gdb/gdb
GNU gdb (Linaro GDB) 7.6-2013.05
Copyright (C) 2013 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=mipsel-openwrt-linux-uclibc".
For bug reporting instructions, please see:
<http://bugs.launchpad.net/gdb-linaro/>.
(gdb) target remote | ssh -T root@respeaker.local gdbserver - python /usr/bin/mopidy --config /root/.config/mopidy/mopidy.conf
Remote debugging using | ssh -T root@respeaker.local gdbserver - python /usr/bin/mopidy --config /root/.config/mopidy/mopidy.conf
Process python created; pid = 27351
stdin/stdout redirected
Remote debugging using stdio
0x77fe0f30 in ?? ()
(gdb) c
Continuing.
warning: GDB can't find the start of the function at 0x77fe0f30.

    GDB is unable to find the start of the function at 0x77fe0f30
and thus can't determine the size of that function's stack frame.
This means that GDB may be unable to access that stack frame, or
the frames below it.
    This problem is most likely caused by an invalid program counter or
stack pointer.
    However, if you think GDB should simply search farther back
from 0x77fe0f30 for code which looks like the beginning of a
function, you can increase the range of the search using the `set
heuristic-fence-post' command.
INFO     Starting Mopidy 2.0.1
INFO     Loading config from builtin defaults
INFO     Loading config from /root/.config/mopidy/mopidy.conf
INFO     Loading config from command line options

Program received signal SIGILL, Illegal instruction.
0x77a93c60 in ?? ()
(gdb) bt
#0  0x77a93c60 in ?? ()
warning: GDB can't find the start of the function at 0x77a93c60.
(gdb)
```

xiongyihui, 2017-09-11:
Try to disable mopidy may solve the issue

```
/etc/init.d/mopidy stop
/etc/init.d/mopidy disable
```

kelnos, 2017-09-11:
@xiongyihui i don't want to disable mopidy; i want to use it.  part of what i'm trying to do is add more extension modules to the existing set.

xiongyihui, 2017-09-12:
It seems when more python modules are used, more chances of segments may happen.

kelnos, 2017-09-12:
@xiongyihui do you know why that is?  that's a *huge* limitation if that's the case, especially if you can't really tell what will cause it to fail without trying a bunch of things.

i mean, even just running `pip install` can trigger a SIGSEGV/SIGILL.  simply "having more modules" shouldn't cause that.

xiongyihui, 2017-09-12:
I've not figure out why :(
When running the same program, sometimes segment fault happens, sometimes not

xiongyihui, 2017-09-15:
@kelnos @Oliv4945 @Noxeus 
Did you use a SD card as extroot when you got the illegal instruction?

kelnos, 2017-09-15:
@xiongyihui Yes.  I've tried the 8GB class-10 card that came with the ReSpeaker Core.  I've also tried two other 32GB class-10 cards (SanDisk).  Same result.

xiongyihui, 2017-09-15:
@kelnos I found that loading large program from SD may trigger `illegal instruction` or `segment fault`.

I've tried to run https://github.com/respeaker/mycroft-core on `/tmp` (it's in RAM) and on `/root/sd1` (it's in SD card with EXT4 format)
1. When the program is on `/tmp`, it almost works well (1 failed with segment fault, 10+ success)
2. When the program is on `/root/sd1`, it failed with `illegal instruction` or 'segment fault` about 4 or 5 times, and then it can work. 

And copy the program from SD card to RAM (`/tmp`), the program is likely to work. It shows the program on SD card is file. 

As openwrt will load programs into RAM before running the program. It seems there is an issue when openwrt loads a large program into RAM.

I also test if it is a problem of reading contents from SD card (via SDIO interface). I've tried to copy large files to the SD card and then check files' MD5. Each MD5 remains the same. I didn't find any errors of reading /writing SD card.

My temporary solution is trying to run a program several times.  For example,

```
while true; do mopidy; done
```

Oliv4945, 2017-09-15:
@xiongyihui : A way to explore but git is not a huge software.
I there any kind of watchdog which can prevent the soft to start if too slow to load ?

Advises from Seeed's tech support, without any success for me.
```
Update for old version

please click here to download the lastest firmware on your computer, copy it to a SD card and plug the SD card into ReSpeaker.

Connect to the serial console of ReSpeaker, type the following command lines to update the firmware:

mount /dev/mmcblk0p1 /mnt
cd /mnt
sysupgrade -n -F ramips-openwrt-latest-LinkIt7688-squashfs-sysupgrade.bin

It will cost about 3 minutes for ReSpeaker to install the firmware and reboot, please don't turn off ReSpeaker when updating.
```
```
Install software on ReSpeaker

After extending storage with a SD card, there are enough storage to install software on ReSpeaker.

Install git

    opkg update
    opkg install git git-http
```
```
Update Python Library

git clone https://github.com/respeaker/respeaker_python_library.git
cd respeaker_python_library
python setup.py install
```

kelnos, 2017-09-15:
@xiongyihui ok, i tried running it in a loop.  i got mopidy to finish initializing (after it crashed 13 times in a row).  i was able to do a music search, and then attempt to start something (via google music).  it failed to play for unrelated reasons... after poking around in mopidy's web UI a bit more, it crashed with another illegal instruction and then took another 4 restarts (heh, so few this time) to get it running again.

this exploration is interesting, but ultimately i don't see a real solution here.

i'm wondering if it's possible to run a more modern kernel and userland on the device.  kernek 3.18.23 is i guess reasonably new (for an old stable series), but uclibc seems dead (no releases in over 5 years), gcc 4.8.3 was released over 3 years ago, python 2.7.9 nearly 3 years ago, and so on.  no guarantee that using newer software will magically fix things, but i'm feeling like it might be worth a try.

xiongyihui, 2017-09-19:
I just made a new firmware with less build-in packages.  It has 10M internal storage space to install extra packages without a SD card. As loading applications from SD card has issues, we can put applications in the internal storage before we figure it out.

[respeaker-20170919.bin.zip](https://github.com/respeaker/get_started_with_respeaker/files/1312892/respeaker-20170919.bin.zip)

### Features
+ juci is removed
+ alexa is added. (run `alexa` to start)
+ some uncommon used gstreamer plugins are removed
+ mopidy webclient is removed. (To install one, try `pip install mopidy-musicbox-webclient`)

To use the new firmware, download the file, extract it and upload the bin file to respeaker.

```
sysupgrade respeaker-20170919.bin
```

kelnos, 2017-09-20:
Ah, cool.  Will try to take a look at it in the next few days.  Interested to see if that helps, though 10MB isn't all that much extra to work with :-/

mctouch, 2017-10-23:
I'm having the same issues, segfaults, loss of ssh, out of memory, tainted mount of /overlay etc. Never did really work for me, constant chase down of errors even with mopidy and many other unnecessary PID's killed. Will try this firmware after a factory reset.

mwie, 2017-10-31:
My problems are the same: segfaults, ... very frustrating! I will try your new firmware, too. Thanks!

jerryyip, 2018-03-28:
@Oliv4945 Hi, we have an new firmware(it was released by @xiongyihui at this issue, few months ago), you should have a try. [click here](https://github.com/respeaker/get_started_with_respeaker/issues/120#issuecomment-330414898)

Oliv4945, 2018-03-28:
@jerryyip : thank you for your answer, I saw it, but also: ` It has 10M internal storage space to install extra packages without a SD card. As loading applications from SD card has issues`
Working without the SD Card is not an option for me given the small space in the internal storage.

Can you please work on the `loading applications from SD card has issues` part ?

Thank you
Olivier

jerryyip, 2018-03-28:
@Oliv4945  what kinds of application you need?

Oliv4945, 2018-03-28:
Hi,

I need at least ` bash git git-http sudo alsamixer aplay arecord whiptail shadow-usermod coreutils-sleep coreutils-pr coreutils-whoami coreutils-groups awk curl iconv jq nano perl sed sox wget` + mp3 files to be played, so i need to use the SD card overlay.
Also I do not even use the ReSpeaker for now because of this issue, I might required more if I am able to use it

jerryyip, 2018-03-28:
Hi @Oliv4945 Due to hardware limitations, Respeaker core is not as powerful as Raspberry, so if you have these advanced needs, Respeaker core v2 is recommended.

closed 2018-04-02 by jerryyip

Oliv4945, 2018-04-16:
Hi @jerryyip,
I am surprised by your answer and the fact that you closed the issue: even `git` or `jq` took alone can give the error, and it is not asking powerful resources. It is listed nowhere on Seeed's website that SD card is for file storage only and we can not load software from it. I would not have bought the Respeaker if I knew this limitation.
The ARM of the core V2 would be easier for me too but I will not spend money on it as I already bought 2 core v1 which are not working as advertised, 1 year after I received them.