respeaker/get_started_with_respeaker #121: Respeaker comport can't detected, Wifi can't detected

Ervanu, 2017-05-09:
I have Respeaker Board v1.0 10/13/2016. When i connect the respeaker board to pc using Mini usb. Comport is undetectable and wifi is also undetectable. Green light on, blue light blinking

Please Help me, how can i solve this problem.

sudhaa1232, 2017-05-11:
do you have respeaker core or mic array.

Ervanu, 2017-05-12:
Thanks for your respone, sudhaa1232 :)
I have both, respeaker core and mic array. But, I just using respeaker core without mic array. Before the problem occurs, the respeaker core can work normally. Alexa and playmusic can running good. I tried to do a program using Arduino IDE.  Com Port detected arduino leonardo. But, i can not uploading program to the respeaker core. There was an error with avrdude. Then, I try to follow the steps on the following link: 
https://github.com/respeaker/get_started_with_respeaker/wiki/SD-card-autorun-script
but in the middle of procces we suddenly got lost connection with the serial monitor. Then the serial (COM port) and wifi connection wasn't exist anymore

xiongyihui, 2017-05-12:
@Ervanu which script did you use? The script to do `firstboot` or the script to re-flash Arduino bootloader?

Ervanu, 2017-05-12:
I used re-flash Arduino bootloader:
  1.  /etc/init.d/mopidy stop  
  2.  /etc/init.d/alexa stop      
  3.  mt7688_pinmux set ephy gpio
  4.  cd /etc/arduino
  5.  avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m  -U 
      flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m
      touch /tmp/run/mountd/mmcblk0p1/done
I didn't put the script on sdcard. I type manually on terminal console. when enter the last script: 

      avrdude -c linuxgpio -p m32u4 -e -U lfuse:w:0xFF:m -U hfuse:w:0xD8:m -U efuse:w:0xCB:m  -U 
      flash:w:Caterina-ReSpeaker.hex -u -U lock:w:0xEF:m
      touch /tmp/run/mountd/mmcblk0p1/done

Suddenly, My Terminal broke up(I'm using Serial terminal), Respeaker wifi wasn't exist, green led was "ON", blue led was "Flashing". @xiongyihui

xiongyihui, 2017-05-12:
You can try the `firstboot` script to bring wifi back.

Ervanu, 2017-05-15:
I have tried firstboot method. I put the script to SDcard without file ending. But that does not solve the problem. Wifi and Serial Port still not detected.

KillingJacky, 2017-10-17:
Please refer to #58 `To fix a totally bricked device`

closed 2017-11-13 by KillingJacky