respeaker/get_started_with_respeaker #125: Use external speaker ?

Oliv4945, 2017-05-13:
Hi,
I tried to connect a 1W 8 OHm external speaker to the Respeaker core and I am not able to get sound on it. `play` and `madplay` are playing on earphones plug only (even if not connected) and `aplay` seems to output on the speaker but is it just white noise.
My `.asoundrc` file:
```
pcm.!default {
  type asym
   playback.pcm {
     type plug 
     slave.pcm "hw:0,0"
   }
   capture.pcm {
     type plug
     slave.pcm "hw:0,0"
   }
}
```
From schematic I assume that on speaker breakout connector, the square is the positive line. Could you please confirm ?

Do you have any idea ?

Thanks

xiongyihui, 2017-05-15:
Have you tried to adjust the volume for the speaker using `alsamixer`?

Oliv4945, 2017-05-15:
>Have you tried to adjust the volume for the speaker using alsamixer?

@xiongyihui.
Yes, I did but it did not helped. Does the Respeaker automatically switch between speakers and headphones if the headphones are not connected ?

Also, could you please confirm the polarity of the breakout connector ?

xiongyihui, 2017-05-15:
When running aplay, both speaker and headphone will  have output. Generally, 1W 8 OHm external speaker has no polarity. I will inform you the polarity of the breakout connector if  I get a response from  our hardware engineer.

closed 2017-10-17 by KillingJacky