respeaker/get_started_with_respeaker #129: Microphone Array sound quality issue

anthonyray, 2017-05-22:
While experimenting with the ReSpeaker, I could not get Snowboy to detect a hotword. 

I tried to record my voice using the `rec` command. 
I'm getting weird results : 
- Most of the times, the recording has a lot of white noise
- Sometimes, it's only the beginning of the recording that has a lot of white noise, while the rest of the record is of OK quality. 

Here are some recordings of me testing the respeaker. They were obtained issuing the `rec` command. 

 -ReSpeaker Test 01 : clyp.it/yjlooz52
- ReSpeaker Test 02 : https://clyp.it/50ouorsz

What could I do to improve this ?

shyhshian, 2017-06-02:
I found mic noisy issue was due to some mics (1 or more of the 7-mics) have mic_data Jitter issue (may be caused by waveform ripple). the mic_data may not be aligned to mic_clock, so XMOS got wrong audio data. I add 58pF capacitor connect from mic_data to GND (you can see, there are spared pads near microphone). the issue solved. For your reference!

xyu6, 2017-11-24:
Hi Dear Customer, please remove C87,C94,C101,C103,C104,c105,c106 capacitors. the noise will be gone. here is the mic array eagle files. you can download the eagle software to open the brd file to find the location. Sorry for the inconvenience. thanks.  
[ReSpeaker Microphone Array BRD](https://github.com/SeeedDocument/ReSpeaker_Mic_Array/raw/master/res/Respeaker%20Microphone%20Array%20v1.0.brd.zip)

closed 2017-12-15 by KillingJacky