respeaker/get_started_with_respeaker #132: Mics array with matlab

eljuligallego, 2017-06-04:
hello,
I would like to know if could be possible to use the mic array with matlab software before purchase this device.

how is this hardware connected and recogniced in a windows/linux enviroment?. like a sound car with 8 input?.

is not, how can I call the hardware to capture the data?.

thanks.

xiongyihui, 2017-06-05:
It depends on matlab. The mic array is a usb sound card.
It seems matlab supports to read multi-channel audio data, see https://cn.mathworks.com/hardware-support/sound-card-daq.html

Generally, we will record a piece of audio with audacity, and then analyze it with matlab. Matlab is some kind of slow for realtime audio processing

eljuligallego, 2017-06-05:
Thanks @xiongyihui , the problem with matlab support for sound cards I have read somewhere that there is a limit of 4 channels so I don´t think this solution gonna work.

The aproach I have in mind is to call a external function to capture the mics data to a file and then read the data from file in Matlab but I would like to make it via scripts to be more fexible and nearest to a real time solution.

In matlab is possible to call external programs via system commands so any command line software for capture sound mics could work but I don´t know if audacity can.

Regards,
Julian.

closed 2017-09-06 by xiongyihui

pmarmaroli, 2019-04-16:
Hello @eljuligallego , I succeeded in interfacing a Respeaker USB 4 Mic Array with Matlab. I didn't find any help on the internet so I share the script:

---------

% init recording parameters
fs      = 16000;
period  = 16000;
nbmic   = 5;

% declare sound card
audioIn = audioDeviceReader(...
    'Device', 'ReSpeaker 4 Mic Array (UAC1.0)  (ReSpeaker 4 Mic Array (UAC1.0))',...
    'Driver', 'DirectSound', ...
    'SampleRate',  fs, ...
    'NumChannels', nbmic ,...
    'OutputDataType','double',...
    'SamplesPerFrame', period,...
    'BitDepth','24-bit integer');

% record signal
frame = audioIn();

% display signals
figure;
for k = 1:nbmic
    hold on
    plot(frame(:,k)-(k-1) )
end

------

For some reason we have to declare 5 channels instead of 4 (I guess first channel is not a mic signal but the result of a processing).

The picture below describes the repartition of each channel.

[image]

Regards,

Patrick