respeaker/get_started_with_respeaker #133: 0x82 Firmware and pixel_ring.py

ncannings, 2017-06-07:
I have previously been able to use the pixel_ring.py and mic_array.py examples to change colours on the mic array leds (and set listen mode etc)

Since upgrading to the 0x82 firmware, this no longer works

Thanks

xiongyihui, 2017-06-07:
The 0x82 firmware is experimental. Have a try - https://github.com/respeaker/mic_array/blob/master/pixel_ring.py

ncannings, 2017-06-08:
I get "No USB device found"

The device is there:  

```
#!/usr/bin/python
import sys
import usb.core
# find USB devices
dev = usb.core.find(find_all=True)
# loop through devices, printing vendor and product ids in decimal and hex
for cfg in dev:
  sys.stdout.write('Decimal VendorID=' + str(cfg.idVendor) + ' & ProductID=' + str(cfg.idProduct) + '\n')
  sys.stdout.write('Hexadecimal VendorID=' + hex(cfg.idVendor) + ' & ProductID=' + hex(cfg.idProduct) + '\n\n')
```
Shows the device in a list

> Decimal VendorID=10374 & ProductID=7
> Hexadecimal VendorID=0x2886 & ProductID=0x7
> 

but this code from the new pixel_ring.py does not seem to find it

  ```
def find(vid=0x2886, pid=0x0007):
        dev = usb.core.find(idVendor=vid, idProduct=pid)
```

susatenk, 2017-06-12:
The problem is not in usb.core.find, pyusb library is old and works fine. I don't know much about python, because I started to do something myself just a couple weeks ago but after today I've tried to control leds and spent day checking it, here is my thoughts. 
Function which belongs to "HID" class and also named "find" returns meaningful value only if both IN and OUT endpoints present. For current firmware of mic_array we have only IN endpoint for Human Interface Device class (0x3) of device config:
```
    INTERFACE 4: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x4
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x1
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x83: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x8
```
Not sure how it was before firmware 0x82, 0x80 is not present on github anymore and I didn't found it in internet so can't check. But I'm expecting 0x80 to have OUT endpoint for HID class populated in device config. Does anybody still have initial RAW firmware and can confirm that? I also checked pixel_ring.py with different XVSM firmware (0x31 and 0x32), bad luck as well. But since RAW firmware is prerequisite to use these python mic array utils, maybe it's expected.

State of support is sad despite it's expected in case of mainstream product. But I'm starting to think 180 dollars investment in Reaspeaker wasn't the wisest decision in my life :)

xiongyihui, 2017-06-13:
Have you tried the firmware -
 https://github.com/respeaker/mic_array_dfu/blob/master/respeaker_mic_array_8ch_raw.bin

susatenk, 2017-06-13:
This firmware is much better! The reported version is the same - 0x82 which is slightly confusing but OUT endpoint present and pixel_ring.py is working now. Thank you!
```
    INTERFACE 4: Human Interface Device ====================
     bLength            :    0x9 (9 bytes)
     bDescriptorType    :    0x4 Interface
     bInterfaceNumber   :    0x4
     bAlternateSetting  :    0x0
     bNumEndpoints      :    0x2
     bInterfaceClass    :    0x3 Human Interface Device
     bInterfaceSubClass :    0x0
     bInterfaceProtocol :    0x0
     iInterface         :    0x0
      ENDPOINT 0x83: Interrupt IN ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :   0x83 IN
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x8
      ENDPOINT 0x2: Interrupt OUT ==========================
       bLength          :    0x7 (7 bytes)
       bDescriptorType  :    0x5 Endpoint
       bEndpointAddress :    0x2 OUT
       bmAttributes     :    0x3 Interrupt
       wMaxPacketSize   :   0x40 (64 bytes)
       bInterval        :    0x1
```

susatenk, 2017-06-13:
So LED control is working in RAW mode only, right?

ncannings, 2017-06-14:
I have checked both - It only works in RAW using the supplied scripts, due to the absence of the OUT endpoint on the XVSM firmware - I have managed a hack, which I will post

@xiongyihui, can we expect an update to the XVSM firmware?

xiongyihui, 2017-06-14:
The pixel_pring.py of  https://github.com/respeaker/respeaker_python_library should work with the XVSM firmware.
Our mic array developer @Fuhua-Chen is out of office. When he come back a few weeks later, he will work on it.