respeaker/get_started_with_respeaker #134: Broken ReSpeaker Mic Array devices (manufacturing issue?)

mattcurf, 2017-06-07:
Two of the four ReSpeaker Mic Array devices received have been broken on arrival. I've downloaded the 8 channel raw FW on each of them to figure out what is happening, and recorded the following 8 channel wave files using the following command: 'arecord -c 8 -r 16000 -t wav test2.wav':
Broken Device 1: https://www.dropbox.com/s/v1v0d5b1rjh6bjo/respeaker_mic_array_device2.wav?dl=0
Broken Device 2: https://www.dropbox.com/s/uyydnxm6f1uas86/respeaker_mic_array_device1.wav?dl=0

In each case, multiple microphones were generating noisy audio streams, or completely broken audio streams, for several of the microphones. This certainly must have been interfering with the DSP's ability to perform proper AEC/beaming/etc.

The other two devices are working as expected.

shyhshian, 2017-06-12:
I found mic noisy issue was due to some mics (1 or more of the 7-mics) have mic_data Jitter issue (may be caused by waveform ripple). the mic_data may not be aligned to mic_clock, so XMOS got wrong audio data. I add 58pF capacitor connect from mic_data to GND (you can see, there are spared pads near microphone). the issue solved. For your reference!

xyu6, 2017-11-24:
Hi Dear Customer, please remove C87,C94,C101,C103,C104,c105,c106 capacitors. the noise will be gone. here is the mic array eagle files. you can download the eagle software to open the brd file to find the location. Sorry for the inconvenience. thanks.  
[ReSpeaker Microphone Array BRD](https://github.com/SeeedDocument/ReSpeaker_Mic_Array/raw/master/res/Respeaker%20Microphone%20Array%20v1.0.brd.zip)

closed 2017-12-15 by KillingJacky