respeaker/get_started_with_respeaker #135: Is Alexa w/ version V2 API ever going to be released?

felixekman, 2017-06-11:
@xiongyihui @yexiaobo-seeedstudio Have you guys stoped this project and left us backers without releasing the promised version V2 that supports the new Alexa apis?

alexeinz, 2017-06-12:
yea whats going on, i see respeaker v2 and all that... ,the original project released and unworkable hardware that hardly even functions as simple airplay speaker... 
if you not gonna develop it further , i want a bloody refund of at least half the cost,because the original promises were all false

xiongyihui, 2017-06-13:
We are trying to bring some open source algorithms (DOA, AEC and etc.)for the mic array at the moment.  It's at https://github.com/respeaker/mic_array
The version V2 is pending. Didn't find any python http2 library which is very suitable to implement the new alexa api yet. [The hyper library](https://github.com/Lukasa/hyper) is a close one, but needs a lot of hack.

@alexeinz Which problem do you have with airplay? When disable other services such as mopidy, the airplay service will be a little more stable.

alexeinz, 2017-06-13:
little more stable , is not stable, it is practically unusable, low sound
quality and often cuts out from multiple sources.
in comparison, almost 10 year old mac express is totally fine.
the product was advertised totally differently originally.
its obvious that it is underpowered for the functionality intended.
instead of concentrating on mic array - how about finish and perfect the
actual product first ?
provide at least a working alexa v2, google voice, real working products
ensure stability and solve existing issues.
or just refund half of the money so we can just write down the product or
go and test your v2?

------
Regards
Alex Einz

Tel - +81(0) 80-6749-1800
dj@alexeinz.info
http://www.alexeinz.info

Weekly quote:
“The true sign of intelligence is not knowledge but imagination
 - Albert Einstein

sevencheng798, 2017-06-14:
@felixekman @alexeinz  I found a good open source hardware that supports alexa - WisCore board, your can try to experience it very great. to share a link: https://github.com/RAKWireless/WisCore/wiki/WisAvs

felixekman, 2017-06-14:
@sevencheng798 Thanks, I've seen your project. But sadly the board in question and the firmware is not compatible with this project. Wrong version of MT7XXX we need it for MT7688. Also, there seems to be other things severely lacking from the platform. For all it's faults the Respeaker is a fine hardware platform, if the makers ( @xiongyihui ) can just get their act together and focus on finishing developing for the already released platform as promised originally.

xiongyihui, 2017-06-14:
Thanks for your feedback. The v2 will got more priority.

closed 2017-09-06 by xiongyihui