respeaker/get_started_with_respeaker #143: WiFi issue with special chars in SSID

tedstriker, 2017-07-12:
I have a closing bracket in my wifi SSID which leads to an error while setting up the connection.
On reboot, ReSpeaker doesn't connect to the wifi and reopens an AP with SSID ReSpeakerA12345

```
root@ReSpeaker:/# wictl 
0, xemxi
1, DIRECT-QT-BRAVIA
2, FRITZ!Box 7362 SL
3, __/)__
Please choose your wifi: 3
Please input the wifi password: xxx
/bin/sh: syntax error: unexpected ")"
udhcpc (v1.23.2) started
Sending discover...
Sending discover...
Sending select for 192.168.1.123...
Lease of 192.168.1.123 obtained, lease time 86400
udhcpc: ifconfig apcli0 192.168.1.123 netmask 255.255.255.0 broadcast 192.168.1.255
udhcpc: setting default routers: 192.168.1.1
success
root@ReSpeaker:/# 
```

joecampbell, 2017-07-14:
As a workaround you can do use the command line and escape special characters
`wictl "ssid" "password"`

tedstriker, 2017-07-14:
Thanks for the work around.
Are you going to fix it in a future firmware?