respeaker/get_started_with_respeaker #145: Security issue: ReSpeakerXXXXX AP doesn't get disabled after configuration

tedstriker, 2017-07-13:
After configuring the regular WiFi access as described, the ReSpeakerXXXXX AP doesn't get deactivated.
So anybody in the neighborhood can connect and change ReSpeakers WiFi config as the like.
Please reconsider this behavior.
Also a workaround is _highly_ appreciated, as well as an updated documentation.

xiongyihui, 2017-07-14:
The "bug" is what we design it to be. AP & STA work at the same time which makes user easier to access the device.

tedstriker, 2017-07-14:
Why do I need to access it through the AP _after_ I set up the connection to my actual wifi network?
People tend to already have their own (secured) wifi networks. You expect that as well, because the landing page is the setup for that wifi connection.
so why would they use that AP after configuration?

What prevents my neighbor from joining the ReSpeakerXXXX AP and, through already implemented traffic forwarding, do what ever he wants in my entire network?

What do you think of the following approach?
Turn on either AP _or_ STA
if STA is unreachable or not configured shut it down and fire up AP. If STA is configured and reachable, shut AP down and fire up STA.

xiongyihui, 2017-07-17:
When the device connected to an AP, it is not easy for most users to find out its IP, so we leave its AP available. You can turn off the device's AP by changing `/etc/config/wireless`.

tedstriker, 2017-07-17:
That sounds good. Could you tell us what to change? I tried it and was out of luck.

xiongyihui, 2017-07-18:
You can remove the `config wifi-iface 'ap'` option to disable the AP

```
config wifi-iface 'ap'
        option device 'radio0'
        option mode 'ap'
        option network 'lan'
        option ifname 'ra0'
        option encryption 'none'
        option ssid 'ReSpeakerXXXX'
        option seq '1'
```

You can also add a password to the AP

```
config wifi-iface 'ap'
        option device 'radio0'
        option mode 'ap'
        option network 'lan'
        option ifname 'ra0'
        option key '{your-password}'
        option encryption 'psk'
        option ssid 'ReSpeakerXXXX'
        option seq '1'
```

After changing `/etc/config/wireless`, reboot to use the new configuration.

tedstriker, 2017-07-18:
Unfortunately this does not work in my setup.
I reset everything before trying your recommendation.

As soon as I remove that part in config respeakers `ra0` interface does not come up automatically after rebooting. If it is brought up manually with `ifconfig ra0 up` it does not obtain an IP.

Did your solution work in your setup?

here's the resulting config:
```
config wifi-device 'radio0'
        option type 'ralink'
        option variant 'mt7628'
        option country 'TW'
        option hwmode '11g'
        option htmode 'HT40'
        option channel 'auto'
        option disabled '0'

config wifi-iface 'sta'
        option device 'radio0'
        option mode 'sta'
        option network 'wan'
        option ifname 'apcli0'
        option led 'mediatek:orange:wifi'
        option ssid 'UplinkAp'
        option key 'SecretKey'
        option encryption 'psk'
        option disabled '1'
        option ApCliEnable '1'
        option ApCliAuthMode 'WPA2PSK'
        option ApCliSsid 'MySSID'
        option ApCliWPAPSK 'MyPASSWORD'
```

tedstriker, 2017-07-24:
@xiongyihui Any updates?

xiongyihui, 2017-07-25:
How about try to run `ifconfig ra0 up` and then `wictl`

The recommended way is to encrypt the respeaker's ap. The configuration is like:

```
config wifi-device 'radio0'
        option type 'ralink'
        option variant 'mt7628'
        option country 'TW'
        option hwmode '11g'
        option htmode 'HT40'
        option channel 'auto'
        option disabled '0'

config wifi-iface 'ap'
        option device 'radio0'
        otion mode 'ap'
        option network 'lan'
        option ifname 'ra0'
        option encryption 'psk'
        option key '{ap-pasword}'
        option ssid 'ReSpeakerXXXXXX'
        option seq '1'

config wifi-iface 'sta'
        option device 'radio0'
        option mode 'sta'
        option network 'wan'
        option ifname 'apcli0'
        option ssid 'UplinkAp'range:wifi'
        option key 'SecretKey'
        option encryption 'psk'
        option disabled '1'
        option ApCliEnable '1'
        option ApCliAuthMode 'WPA2PSK'
        option ApCliSsid 'AP'
        option ApCliWPAPSK '12345678'

```