respeaker/get_started_with_respeaker #146: web terminal RSA key

yyd19948, 2017-07-14:
web terminal上出现IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!                                                                               Someone could be eavesdropping on you right now (man-in-the-middle attack)!                                                         It is also possible that a host key has just been changed.                                                                          The fingerprint for the RSA key sent by the remote host is                                                                          SHA256:cw55L/zy8KdAYnPdz9fKdTY5ZJKR/yPG+ZYkI1q+9mg.                                                                                 Please contact your system administrator.                                                                                           Add correct host key in /root/.ssh/known_hosts to get rid of this message.                                                          Offending RSA key in /root/.ssh/known_hosts:1                                                                                       RSA host key for localhost has changed and you have requested strict checking.                                                      Host key verification failed.       /root/.ssh/known_hosts:1  只可读不可写

xiongyihui, 2017-07-14:
如果用SD卡做了extroot，然后重置了系统，SD的内容并不会被清楚掉，还要手动格式化SD卡。

出现这种情况，可以格式化SD卡，或者通过usb虚拟串口登录进去删掉 /root/.ssh

yyd19948, 2017-07-14:
sd卡没有做extroot，只是第一次进入web terminal时设置了一个密码，密码忘记了，输了几次都不对就出现这种情况了

tedstriker, 2017-07-14:
do you mind to repeat your answer in English?

closed 2017-09-06 by xiongyihui