respeaker/get_started_with_respeaker #152: Does Respeaker Mic Array work with Windows 10 IoT Core on the Raspberry Pi?

alexschneider, 2017-07-21:
Hi,

I'm looking at using the respeaker mic array for a project that requires a UWP windows application. Has there been any testing with the mic array on a raspberry pi running iot core? 

Thanks!

haplo10, 2017-07-31:
Hi,
i'm looking the same, the Raspberry windows 10 haven't driver for respeaker, i tried to connect but i wasn't able to enable it.
If you find something new please tell me.

thanks,

Massimo

xiongyihui, 2017-08-01:
Not available yet. We are working on a UAC 1.0 firmware. Hope it can be used on Windows 10 IoT

closed 2017-09-06 by xiongyihui

abhijitdalavi, 2017-10-16:
Hi,
i'm looking the same, any updates on windows iot core devices support with Respeaker?
If you find something new please tell me.

thanks,

haplo10, 2017-10-16:
Hi,
no news, i hope to find something in the next months...
if i know news i'll tell you, do it the same for me.

bye.

abhijitdalavi, 2017-11-27:
hi guys ,
I have tried to sucessfully install the respeaker windows driver using  powershell on windows 10 iot core reaspberry pie.
i have attached screenshot below ....what you thing will it work on windows 10 iot core??
[image]

haplo10, 2017-11-29:
Hi...
i try your procedure and insatallation process has no errors but i see only Respeaker speaker working, microphone isn't detected.
Is the same for you? any ideas?

thanks.
Massimo

abhijitdalavi, 2017-11-29:
Hi Massimo,
Have you tried to increase microphone settings for volume in...windows
device portal ..may be microphone sensitivity is low that will also be a
cause
Thanks
Abhijit

haplo10, 2017-11-29:
I have slide disabled....

[image]

abhijitdalavi, 2017-11-29:
Try to restart and then check ...if it shows like previous one..

haplo10, 2017-11-29:
just tried....

haplo10, 2017-11-29:
[image]

no mic device...

abhijitdalavi, 2017-11-30:
My be the issue was same
https://github.com/Fuhua-Chen/ReSpeaker_Microphone_Array_Driver/issues/2

See this site ...
https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/peripheraldrivers
Get the updated drivers
Try to reinstall and check again

abhijitdalavi, 2017-11-30:
TRY THIS UPDATED RESPEAKER DRIVER
https://github.com/Fuhua-Chen/ReSpeaker_Microphone_Array_Driver

haplo10, 2017-12-04:
nope....
i tried to uninstall driver and than use driver from your link but nothing....

abhijitdalavi, 2017-12-04:
which you have used UAC 1.0 OR UAC 2.0?

abhijitdalavi, 2017-12-04:
Usb audio class driver 1.0 or 2.0

haplo10, 2017-12-04:
[image]

abhijitdalavi, 2017-12-05:
Hi,
As I said you have to try ...the universal .Inf files cause windows 10 IoT
core supports only universal .inf ...please get the universal .inf from
your updated windows 10 pc ....from c:/windows/inf
The file is usbaudio2.inf...
Try to install  using
https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/peripheraldrivers
this link ...this may help

abhijitdalavi, 2017-12-05:
http://www.xmos.com/support/usb-audio-driver-support
If that not worked try these drivers

haplo10, 2017-12-11:
i downloaded all tools to try what you told me, but i'am not able to create package, it return always an error.

Badger1989, 2018-08-08:
Has anyone managed to get the mics to work correctly on Win 10 IoT. I see it listed correctly and can edit both the speaker and mic volume controls. The issue I have is in my test app when I record an audio sample i just get noise. If I clap I can just about make out the clap due to the noise changing in volume level. Speech can not be made out at all.  If I run the same app on my PC i get recorded audio as expected. 

The speaker works fine on both the Pi 3B+ and the pc. 

Looking around the net I cant find a single other post relating to getting this working on Win 10 IoT.

endyb66, 2018-08-30:
I'm also interested on this. For my project i also need the respeaker array on win 10 iot core. So it would be very cool, if this cohld work and if so, it would be nice, when someone could write a small guide.

rajdpandey, 2019-03-16:
Has Anybody Got this Problem Solved?