respeaker/get_started_with_respeaker #153: Bing speech to text

Klervix, 2017-07-24:
Hi, 
I obtained : 
`requests.packages.urllib3.connectionpool:"POST /token/issueToken HTTP/1.1" 401 143`
while trying to do the "Respeaker, play music!".
A lot of endpoints of the MS-Azure API changed, and I think this is the problem here. 
Could you please tell me where the SDK folder is located in order to update the endpoint?

Thank you,
Klervi

devildog125, 2017-07-26:
I've got the same issue as well

xyu6, 2017-11-24:
please use [bing_speech_api.py](https://github.com/respeaker/respeaker_python_library/blob/master/respeaker/bing_speech_api.py) . please refer to Bing Speech API not working #13 for more detail info. thanks. Bill

KillingJacky, 2017-12-15:
Will close this issue, please open a new issue if you have further questions.

closed 2017-12-15 by KillingJacky

cwcentral, 2019-06-21:
Did the endpoints change again. Cause I cannot run the basic main() in bing_speech_api.py.
After registering for Bing STS, I notice the endpoint is: https://westus.api.cognitive.microsoft.com/sts/v1.0

which appears different from the code.

DEBUG:urllib3.connectionpool:https://api.cognitive.microsoft.com:443 "POST /sts/v1.0/issueToken HTTP/1.1" 401 145
Could not request results from Microsoft Bing Voice Recognition service; http request error with status code 401