respeaker/get_started_with_respeaker #156: Is Respeaker "Core" needed to try long distance?

umezy, 2017-07-27:
Hi! now I use mic_array_8ch dfu by ReSpeaker Mic Array.
But I think it is not enough to use in long distance voice capture.
In this purpose, Is Respeaker "Core" needed?
I only have ReSpeaker Mic Array.
thanks!

xiongyihui, 2017-09-06:
If you want long distance, you should use origin firmware

piLinux, 2017-11-09:
For voice capture, you don't need the core. You can attach the mic array to any USB device (it could be with your laptop or a raspberry pi).

For windows you need to install drivers [ [Link](https://github.com/Fuhua-Chen/ReSpeaker_Microphone_Array_Driver/blob/master/ReSpeakerMicrophoneArrayDriver-20171017.zip) ]
For linux you don't need any additional drivers.

closed 2017-11-20 by xiongyihui