respeaker/get_started_with_respeaker #158: How to interface ReSpeaker with a standard USB 2.0 port?

jmbv2015, 2017-08-12:
Is this possible to input sound to any Windows application?

xiongyihui, 2017-08-13:
Which hardware do you use? the mic array or the repseaker core?

The mic array is a USB sound card. You can access it through windows audio api, or use an audio library like portaudio.

The respeaker core has a USB virtual serial port.

jmbv2015, 2017-08-13:
Thank you so much for your immediate answer. We are impressed about this
type of customer support!

To fully understand your questions and being able to address them, let me
please explain in detail what we would like to do with your impressive
array microphone: high-quality recording of business meetings, when people
are sitting around a table, with a Windows or Macintosh computer and any
suitable application --as for instance Audacity for Microsoft Windows.

With the term "high-quality recording" we mean to be able to get a file of
at least 16 bits, Mono, 11 khz --or better, and without too much background
noise.

The question are:

1. How do you think we should proceed to get the best possible results?
Could you please provide all necessary details of your recommended setup?

2. If the results are suitable for our purposes, how could we purchase the
necessary ReSpeaker elements from you at the best possible terms and
conditions? Could you send us a price list for quantities?

JM Boccio
SpeechWare

jmbv2015, 2017-08-13:
Thank you so much for your immediate answer. We are impressed about this
type of customer support!

To fully understand your questions and being able to address them, let me
please explain in detail what we would like to do with your impressive
array microphone: high-quality recording of business meetings, when people
are sitting around a table, with a Windows or Macintosh computer and any
suitable application --as for instance Audacity for Microsoft Windows.

With the term "high-quality recording" we mean to be able to get a file of
at least 16 bits, Mono, 11 khz --or better, and without too much background
noise.

The question are:

1. How do you think we should proceed to get the best possible results?
Could you please provide all necessary details of your recommended setup?

2. If the results are suitable for our purposes, how could we purchase the
necessary ReSpeaker elements from you at the best possible terms and
conditions? Could you send us a price list for quantities?

xiongyihui, 2017-08-14:
[The respeaker mic array](https://www.seeedstudio.com/ReSpeaker-Mic-Array-Far-field-w--7-PDM-Microphones--p-2719.html) is designed for the case. It has 6+1 PDM microphones and built-in Noise Suppression, Beamforming algorithms. It outputs 2 channels 32k/16k audio.

jmbv2015, 2017-08-14:
Perfect: we are going to have supported this project in such a case! However, the main question persists, can you outline how to do it? Can you answer the questions that we are specifically putting forward ? If not, who can we contact to? We are ready to pay for such an advice if needed...

jmbv2015, 2019-02-21:
WE WOULD LIKE TO RETAKE THIS PROJECT BUT NEVER RECEIVED A PROPER ANSWER TO OUR PREVIOUS QUESTION ON THE 14TH AUGUST 2017. ALSO, WHAT IS YOUR DIRECT EMAIL AND PHONE? WHAT IS THE PRICE OR RESPEAKER FOR QUANTITIES?

jmbv2015, 2019-02-21:
WHAT WE REALLY NEED IS AN USB DRIVER FOR WINDOWS. Otherwise, we will not be able to evaluate your  equipment for our purposes...

xiongyihui, 2019-02-22:
You can get it from https://www.seeedstudio.com/ReSpeaker-Mic-Array-v2-0-p-3053.html
It is a standard USB sound card supported by Windows 10.

xiongyihui, 2019-02-22:
@xyu6

jmbv2015, 2019-02-22:
Thank you for your immediate answer, although you are not sending us your email address for easier communication. Whatever the case, we still have the following questions: (1) Do you also need the ReSpeaker Core for the device to work as a desktop USB microphone with Windows 10? (2) How could we immediately order a brand new version 2 of ReSpeaker and, eventually, version 2 of its Core if it exist or we need it (we have already the original Core unit), BUT FIRMWARE UPGRADED AND FINE TUNED FOR BEST PERFORMANCE WITH SPEECH RECOGNITION AND WINDOWS 10? We do not have the expertise neither the time to do this and we do not mind to pay for your work.

jmbv2015, 2019-02-22:
...by the way what is the meaning of @xyu6?

xiongyihui, 2019-02-22:
For your questions:
(1) The USB Mic Array can work without the ReSpeaker Core on Windows 10
(2) You can directly order from https://www.seeedstudio.com/ReSpeaker-Mic-Array-v2-0-p-3053.html. The mic array is able to plug-n-play on Windows 10. For the firmware upgrade and the speech recognition, I would invite @xyu6  to the conversation. @xyu6  is the leader of our support team.

jmbv2015, 2019-02-22:
Thanks!
(1) What is the ReSpeaker Core then for? Is version 1.0 of USB Mic Array (the one we currently have) compatible with Windows 10?
(2) Is it the COMPLEX firmware upgrade really necessary for speech recognition on Windows? Can @xyu6 share information and tips on getting the maximum performance of your USB Mic Array with speech recognition on Windows, which needs the maximum level of ambient noise rejection?

jmbv2015, 2019-02-23:
WE CAN NOT STILL FIND THE WINDOWS 10 COMPATIBLE AUDIO DRIVERS FOR RESPEAKER MIC ARRAY VERSION 1. PLEASE PROVIDE THE CORRECT LINK. THE ONE WE ARE USING DOES NOT WORK!

xyu6, 2019-02-26:
Hi there, 

> (1) What is the ReSpeaker Core then for? Is version 1.0 of USB Mic Array (the one we currently have) compatible with Windows 10?

A. The respeaker core v1.0 (MT7688 chip) runs the Linux based OpenWrt,  So you can pick the voice from respeaker mic array v1.0 and do the voice interaction with alexa or use the bing stt to control devices. 

B. It is compatible with win10, but you need to install the driver, please try to install https://github.com/Fuhua-Chen/ReSpeaker_Microphone_Array_Driver/raw/master/ReSpeakerMicrophoneArrayDriver-20171017.zip and let us know if it works. 

> (2) Is it the COMPLEX firmware upgrade really necessary for speech recognition on Windows? Can @xyu6 share information and tips on getting the maximum performance of your USB Mic Array with speech recognition on Windows, which needs the maximum level of ambient noise rejection?

A. We have 2 firmware, one is raw and the other is DSP processed. You can refer to https://github.com/Fuhua-Chen/ReSpeaker_Microphone_Array_Firmware, Coz the respeaker mic array v1.0 is end of life. There will be no firmware update. thanks for understanding. 

Please drop mail to techsupport@seeed.cc if you have any technical issue. thanks. 

best rgds
Bill

jmbv2015, 2019-02-26:
This is very thorough and we do appreciate it very much!

 

 

JM Boccio

CEO

 

 

SPEECHWARE

The multilingual one stop Web shop 
for Digital dictation & Speech recognition

Square Ambiorix 30 – 5th floor

1000 Brussels (Belgium)

+32 2 7331687

+34 91 1234 635 (from Spain)

 <http://www.speechware.be/> www.speechware.be

 <http://www.quicktypist.com/> www.quicktypist.com

 <http://www.flexymike.com/> www.flexymike.com 

 <http://www.tablemike.com/> www.tablemike.com

 <http://www.digalaw.com/> www.digalaw.com

Twitter: @DigaLaw

Enviado el: Tuesday, February 26, 2019 11:41 AM
Para: respeaker/get_started_with_respeaker <get_started_with_respeaker@noreply.github.com>

 

Please drop mail to techsupport@seeed.cc <mailto:techsupport@seeed.cc>  if you have any technical issue. thanks.

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus