respeaker/get_started_with_respeaker #159: raspberry pi is not picking up respeaker as a mic

Mn0491, 2017-08-22:
I plugged respeaker into the rapsberry pi through the usb port. What are configurations I need to get it working? 

more info

```
lsusb
Bus 001 Device 008: ID 2886:0007  </br>
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: UAC20 [ReSpeaker MicArray UAC2.0], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

Thanks in advanced

xiongyihui, 2017-08-23:
You can record audio using `arecord -v -D plughw:1 -f S16_LE -r 16000 -c 2  audio-2ch-16k.wav`

`-D plughw:1` is to specify the hardware - respeaker mic array.

Mn0491, 2017-08-23:
Is there no way to just make the respeaker the default mic? Or do i have to use that command everytime I want to record audio @xiongyihui

xiongyihui, 2017-08-23:
You add  a alsa configuration file `~/.asoundrc`with contents:

```
pcm.!default {
    type plug
    slave.pcm "hw:1"
}
```

Mn0491, 2017-08-23:
@xiongyihui changing the `~/.asoundrc` to the contents above did nothing and the command
 `arecord -v -D plughw:1 -f S16_LE -r 16000 -c 2 audio-2ch-16k.wav` gives me this error message

```
Recording WAVE 'audio-2ch-16k.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
ALSA lib pcm_params.c:2162:(snd1_pcm_hw_refine_slave) Slave PCM not usable
arecord: set_params:1204: Broken configuration for this PCM: no configurations available
```

Atleast we are getting an error message now! getting somewhere :). do you have any more tips to troubleshoot this?

Mn0491, 2017-08-23:
So i switched to use another respeaker mic array and the command `arecord -v -D plughw:1 -f S16_LE -r 16000 -c 2 audio-2ch-16k.wav` works! I get a wav recording after doing a keyboard interrupt. But the wav file has a bunch of white noise, an overwhelming amount. I can barely hear me speaking in the wav file due to the white noise. 

I've used this to update my respeaker firmware (https://github.com/respeaker/mic_array_dfu) before trying any of this. 

Also when speaking the led does not light up in any direction. May be a hardware issue?

Mn0491, 2017-08-23:
More debugging info. when i play the wav file using headphones, the right side is white noise (static noise) and the left side has a clear recording.

xiongyihui, 2017-08-24:
if you use new firmware, try to record 8 channel audio instead of 2.

```
arecord -v -D plughw:1 -f S16_LE -r 16000 -c 8 audio-8ch-16k.wav
```

closed 2017-09-06 by xiongyihui