respeaker/get_started_with_respeaker #161: included compiler toolchain is broken

kelnos, 2017-08-24:
Not sure where to file issues about this sort of thing; I don't see a ReSpeaker forum or something, and the other repos didn't seem appropriate.

I'm trying to install the Google Music extension to Mopidy, which failed due to a compiler crash.  It turns out that there seems to be something wrong with the compiler, even a "hello world" app fails:

```
root@ReSpeaker:~# cat hello.c 
#include <stdio.h>

int main(int argc, char **argv) {
    printf("hello world\n");
    return 0;
}
root@ReSpeaker:~# mipsel-openwrt-linux-uclibc-gcc hello.c -o hello
<built-in>:0:0: internal compiler error: Illegal instruction
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://dev.openwrt.org/> for instructions.
```

It's not completely deterministic: running it over and over causes the illegal instruction to happen at various points, and sometimes it segfaults:

```
root@ReSpeaker:~# mipsel-openwrt-linux-uclibc-gcc hello.c -o hello
hello.c:1:0: internal compiler error: Segmentation fault
 #include <stdio.h>
 ^
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
mmap: Invalid argument
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://dev.openwrt.org/> for instructions.
```

Any ideas?

xiongyihui, 2017-08-24:
You'd better cross-compile the extension. Never tried to compile program on openwrt

kelnos, 2017-08-24:
@xiongyihui unfortunately, cross-compiling python extensions seems to be a bit of a black art, and so far none of the tools i've tried work.  i'm in the process of trying to build a chroot to build inside of (using qemu), but not sure if that'll work either.  do you know of any pointers/resources to help with this?

xiongyihui, 2017-08-25:
We have cross-compiled a few python extensions with c, such as pycrypto and psutil in https://github.com/respeaker/respeaker-feed

There is a guide to build an openwrt package, see https://wiki.openwrt.org/doc/howtobuild/single.package

kelnos, 2017-08-25:
I've been trying to get the single-package build instructions working, but it's been remarkably difficult to get even the base toolchain to build.  After fixing a bunch of issues I'm currently stuck on the inability of gcc 7.2.0 (my host CPU's compiler) to build linaro 4.8 (gcc 7.x got more strict about some things).

I'll take a look at the respeaker-feed repo; thanks for the pointer!

closed 2017-08-25 by kelnos

theronic, 2019-02-10:
How to cross compile a C hello world that can run on Respeaker?

theronic, 2019-02-12:
I've spent two days on this. I can't get ReSpeaker to compile a hello world C program.

kelnos, 2019-02-12:
@theronic my assessment is the v1 hardware has serious stability flaws.  The dev team appears to have moved on to v2, so I wouldn't expect any support.

xiongyihui, 2019-02-13:
@theronic To compile a C program for respeaker v1 is a little hard, as a cross compile environment is required. There are some documents of building and creating a package:

https://openwrt.org/docs/guide-developer/build.a.package
https://openwrt.org/docs/guide-developer/packages