respeaker/get_started_with_respeaker #162: Stackable speaker?

crazyquark, 2017-08-29:
I love the stackable mic array on top of respeaker but having to attach a speaker using the audio jack ruins the form factor.

Is there a way to stack a speaker on top of the mic array, similar to how the mic array stacks on top of the core?

Like a 3 layered cake :) ?

xiongyihui, 2017-08-30:
Yes,  you just need to connect the headphone pins `HP_L` and `HP_R`

[image]

See also https://github.com/respeaker/get_started_with_respeaker/blob/master/Introduction.md#hardware

crazyquark, 2017-08-30:
That's nice but I was wondering if it can also attached to the Mic Array?

xiongyihui, 2017-08-30:
The location of the HP_L and HP_R is the same on the mic array.

crazyquark, 2017-08-30:
All right, thank you. Any recommended speaker IC to use for this?

closed 2017-08-31 by crazyquark