respeaker/get_started_with_respeaker #163: how to record single channel (mono) by the mic array?

giggslam, 2017-09-04:
Hi all, i tried to use mic array in Raspberry pi, and connect Google cloud speech api. However, I found Google speech api only support mono sound (1 channel). 
I tried to record into mono, but not work.

arecord -v -D plughw:1 -f S16_LE -r 16000 -c 1 audio-1ch-16k.wav

Anyone can help?
Thanks a lot!

xiongyihui, 2017-09-04:
What's the output of the command?

closed 2017-09-06 by xiongyihui

susatenk, 2018-05-02:
Hi! The same issue. 

arecord -v -D plughw:1 -f S16_LE -r 16000 -c 1 audio-1ch-16k.wav
During execution no errors, file generated OK but contains just silence, nothing captured. 

arecord -v -D plughw:1 -f S16_LE -r 16000 -c 2 audio-2ch-16k.wav
In case of two channels wav recorded fine.

My asound.conf:
pcm.!default {
  type asym
  playback.pcm {
     type plug
     slave.pcm "hw:0,0"
  }
  capture.pcm {
     type plug
     slave.pcm "hw:1,0"
  }
}

Any advice appreciated. I have dropped experiments with Respeaker Mic Array a year ago, almost immediately after buy, but recently decided to revive my home project.  And stuck again on the same step.