respeaker/get_started_with_respeaker #165: Change Mic array output data to single channel

AhQi, 2017-09-08:
Hi. When I try to record with mic array through "arecord -f S16_LE -D plughw:1 -d 5 -c 1 -r 16000 test.wav", It generated the wav file without any sound. But it worked with input argument "-c 2", is there any way to make single channel output available?

xiongyihui, 2017-09-12:
Which OS did you use? You can use sox to convert stereo to mono. For example
```
sox infile.wav outfile.l.wav remix 1
```

See https://www.nesono.com/node/275

xiongyihui, 2017-09-12:
It's weird. I can record mono audio with arecord

AhQi, 2017-09-12:
Hi, thanks for reply. I solve it with reference of [http://nairobi-embedded.org/alsa_route_and_volume_plugin.html]. Add some config in asound.conf .Mix 2 input channel into 1 output channel.

closed 2017-09-15 by xiongyihui