respeaker/get_started_with_respeaker #166: Can I receive the angle and string(Text) from ReSpeaker Core?

pine3825, 2017-09-17:
Dear

Can I receive the angle and string(Text) from ReSpeaker Core when I say the  wakeup word?
(On Linux)
How can I do that? Please let me know the process.

I have the audio leonardo board and Up Board.
Can I connect the Respeaker to Up Board by USB or UART??

xiongyihui, 2017-09-20:
You can use https://github.com/respeaker/respeaker-xmos-hid

pine3825, 2017-09-28:
I cloned the respeaker-xmos-hid on my UP Board connected by USB ReSpeaker Mic Array
but there is error.
Could you help me?
------------------------------------
jaehoon@jaehoon-UP:~/Workspace/ReSpeaker/respeaker-xmos-hid$ python listen_and_get_position.py 
Traceback (most recent call last):
  File "listen_and_get_position.py", line 2, in <module>
    import hid # https://pypi.python.org/pypi/hidapi aka https://github.com/trezor/cython-hidapi
ImportError: No module named hid

xyu6, 2017-10-20:
If you only use the respeaker core, you can not get the angle. coz it only has 1 mic. if you use the mic array, we can get the angle. which firmware do you use? thanks. Bill

closed 2017-11-13 by KillingJacky